osrf/subt #810: Breadcrubs failure in Tunnel section of Finals Qualification world

m3d, 2021-03-08:
I am observing strange behavior of breadcrumbs in tunnel section of finals qualification world (going straight on the first crossing). Yesterday I run experimentally 4x Freyja robots (one to left/Urban, one to right/Cave, and two straight/Tunnel), and if you see report from Teambase how are they went, you can tell, that in tunnel is something wrong:
[image]
The Cave robot turned up-side-down shortly after entering the section, so ignore the green. But note the difference of orange robot in Urban part deploying only 6 breadcrumbs and blue/red trace of tunnel (with up-to 12 breadcrumbs from two robots).

The D-robot went all the way to Survivor and Drill, but on the return fall off the vertical shaft as A-robot. The reports of artifact and positions is sent regularly.

I think I am seeing similar problem reported to `subt-help` (Nov 27, 2020) in "Tunnel Circuit World 4" where the breadcrumbs were not helping in long straight tunnel??

m3d, 2021-03-08:
Here is older run of two Freyja robots exploring the central part and returned successfully home (and thus Teambase has complete trajectory and artifacts):
[image]

The density of breadcrumbs you can see on video
https://youtu.be/TuSeRfr-DaU

Please let me know what info could help to debug this issue?

m3d, 2021-03-08:
p.s. I wonder if this could be related to "mix of breadcrumbs from different robots"? Now I am running 3x Freyja (left, center, right) and it works fine and scores as it sees new artifact:
[image]
Also the position of Teambase is slightly different (I think I used the 1st slot this time) and robot could drop breadcrumbs +/- 1m in different location??

m3d, 2021-03-08:
one more example without breadcrumbs (X4 drone)
[image]
so it looks like the signal ends in the bend ... so in the 4-robots run it was similar to "no breadcrumbs" situation even there were some deployed. Note that "red trace" in the original example was also short (like for the drone) and blue was slightly longer (the first deployed breadcrumb corresponded to "blue robot")??

m3d, 2021-03-10:
I tried to run two independent runs where two robots enter the Tunnel section and so far only the first one (the one which deployed the breadcrumb in the first turn) is in touch with Teambase all the time ...

Teambase info:
[image]

A-robot (the first one, dropping the "common" breadcrumb)
[image]

B-robot, using shared breadcrumbs and loosing contact with Teambase
[image]

p.s. I may try to create new images and reduce the shared traces to only "online" packets reported by robots at their current position.

acschang, 2021-03-10:
> p.s. I wonder if this could be related to "mix of breadcrumbs from different robots"?
Every breadcrumb dropped affects the communication ability of every robot (and every robot type) so dropping breadcrumbs from different vehicles in a heterogeneous swarm should not have a negative effect or limit communication in any way.

We appreciate you looking into this issue and it is apparent from your results and our internal testing that the level of communication possible in Tunnel Circuit tiles is not on par with communication in Urban and Cave Circuit tiles.  We will pursue avenues for bridging the communication discrepancy between the three subdomains.

m3d, 2021-03-10:
@acschang thanks - there could be some other reason, so I am still not sure about the reason. Below are two samples and it is strange that one robot does not drop breadcrumbs
[image]
[image]

acschang, 2021-03-11:
> @acschang thanks - there could be some other reason, so I am still not sure about the root cause. Below are two samples and it is strange that one robot does not drop breadcrumbs

The case you're reporting is definitely strange however we have had instances of this reported before (#655).  We're working on addressing the issue and adding an avenue for breadcrumb deployment feedback (#814).

caguero, 2021-03-26:
@m3d , any chance you could try [this branch](https://github.com/osrf/subt/compare/issue_810) and reproduce your tests to see if you observe better results in the tunnel section? Thanks!

m3d, 2021-03-28:
@caguero - thanks for the experiment, unfortunately locally I am not able to run two robots with teambase yet and on CloudSim I cannot choose the branch, so the probability of test in near future is low, sorry :( [I will discuss this with my colleagues]

caguero, 2021-04-01:
I did some experiments with the default version (not with the suggested branch for testing) and the results look reasonable to me. See the animated gif attached. 

[image]

The initial conditions are two robots on the staging area. The first visualization shows the area covered without deploying any breadcrumbs. 

Then, I move forward one robot, pass the tunnel intersection and deploy one breadcrumb in the middle section of the straight tile. The second visualization shows how the comms range extends to one extra tile. 

Finally, I keep moving the robot until the next intersection and deploy another breadcrumb. The comms range extendes a few extra tiles. Note that the comms visualization is always from the perspective of the robot sitting in the staging area.

Could you try to reproduce my experiment? I can also try to reproduce your setup if you provide a minimum experiment that seems strange (ideally with two robots and the poses where you deployed the breadcrumbs).

m3d, 2021-04-02:
@caguero thanks for the test and nice presentation - it looked to me that the problem is really combination of TWO robots where some breadcrumbs should be shared. In our implementation we share global positions of all deployed breadcrumbs and place them spatially (30 meters spacing at the moment). So maybe I could add logging of strength or what you would suggest? It looks like that only one of the robot (the one which placed the nearest breadcrumb in the first turn of tunnel section) is later able to talk to the Teambase.

caguero, 2021-04-02:
Could you post a sketch of where the two robots are and where the breadcrumbs are located?

m3d, 2021-04-02:
> Could you post a sketch of where the two robots are and where the breadcrumbs are located?

probably the best is the "ground truth" from https://github.com/osrf/subt/issues/810#issuecomment-795980251

caguero, 2021-04-02:
Do you have the poses of the breadcrumbs and robots by any chance?

caguero, 2021-04-02:
> > Could you post a sketch of where the two robots are and where the breadcrumbs are located?
> 
> probably the best is the "ground truth" from [#810 (comment)](https://github.com/osrf/subt/issues/810#issuecomment-795980251)
> (breadcrumbs are the pink triangles)

And how are the robots represented in the diagram?

zwn, 2021-04-02:
Breadcumbs are the pink downward facing triangles. There are 2 robots represented by their trajectories and the trajectories are the same color as their names in the legend down left. It is a 2D rendering of state.tlog file after the run finishes.

m3d, 2021-04-02:
Uhh, it has been a while ... the logs are on CloudSim, just to find the relevant - I will try to look it up. The robots are represented by their colored trajectory.

m3d, 2021-04-02:
> Do you have the poses of the breadcrumbs and robots by any chance?

The first simulation was `ver99octofqr1-81d7812a-7a4a-47cd-b7f8-b8f47c0a4370` with
```
A210LXM__breadcrumb___0: 29.54 0.57 2.90
A210LXM__breadcrumb___1: 51.68 -19.57 5.00
A210LXM__breadcrumb___2: 81.66 -18.82 5.00
A210LXM__breadcrumb___3: 104.50 0.65 5.00
A210LXM__breadcrumb___4: 134.22 0.62 0.67
```

The second simulation was `ver99octofqbread1-3e9df022-e221-45de-8a5f-a67f8821920f` with
```
B120CXM__breadcrumb___0: 29.54 -0.18 2.90
B120CXM__breadcrumb___1: 48.18 -23.50 5.00
B120CXM__breadcrumb___2: 47.93 -43.55 0.00
B120CXM__breadcrumb___3: 47.81 -49.98 0.00
B120CXM__breadcrumb___4: 48.71 -55.65 0.00
B120CXM__breadcrumb___5: 52.43 -59.19 0.00
A60W210LXM__breadcrumb___0: 87.56 -8.27 5.00
A60W210LXM__breadcrumb___1: 116.08 0.60 5.00
```

caguero, 2021-04-02:
> The first simulation was `ver99octofqr1-81d7812a-7a4a-47cd-b7f8-b8f47c0a4370` with
>
> ```
> A210LXM__breadcrumb___0: 29.54 0.57 2.90
> A210LXM__breadcrumb___1: 51.68 -19.57 5.00
> A210LXM__breadcrumb___2: [...]

This is great, could you also get the robot poses? This is critical to reproduce your experiment.

caguero, 2021-04-02:
> > > Do you have the poses of the breadcrumbs and robots by any chance?
> > 
> > 
> > The first simulation was `ver99octofqr1-81d7812a-7a4a-47cd-b7f8-b8f47c0a4370` with
> > ```
> > A210LXM__breadcrumb___0: 29.54 0.57 2.90
> > A210LXM__breadcrumb___1: 51.68 -19.57 5.00
> > A210LXM__breadcrumb___2: 81.66 -18.82 5.00
> > A210LXM__breadcrumb___3: 104.50 0.65 5.00
> > A210LXM__breadcrumb___4: 134.22 0.62 0.67
> > ```
> > 
> > 
> > The second simulation was `ver99octofqbread1-3e9df022-e221-45de-8a5f-a67f8821920f` with
> > ```
> > B120CXM__breadcrumb___0: 29.54 -0.18 2.90
> > B120CXM__breadcrumb___1: 48.18 -23.50 5.00
> > B120CXM__breadcrumb___2: 47.93 -43.55 0.00
> > B120CXM__breadcrumb___3: 47.81 -49.98 0.00
> > B120CXM__breadcrumb___4: 48.71 -55.65 0.00
> > B120CXM__breadcrumb___5: 52.43 -59.19 0.00
> > A60W210LXM__breadcrumb___0: 87.56 -8.27 5.00
> > A60W210LXM__breadcrumb___1: 116.08 0.60 5.00
> > ```
> 
> This is great, could you also get the robot poses at the moment where the comms are strange? This is critical to reproduce your experiment.

m3d, 2021-04-02:
Well the robots are moving and they are reporting its trajectory. And in the second case you can see that A robot waited 60s and then followed the **left** wall. The common part was placed by robot B, starting immediately and moving in the center (and later falling down from vertical shaft.
[image]
And A robot is no longer visible from the Teambase.

m3d, 2021-04-02:
p.s. maybe just checking - the breadcrumbs can be shared via all robots, right?

caguero, 2021-04-02:
When a robot drops a breadcrumb, it affects all robots and breadcrumbs in its vicinity. Is that what you mean?

m3d, 2021-04-02:
OK, I just wanted to check that really two breadcrumbs dropped by orange robot B should help also the blue robot A ... OK, then I still think that there is some bug and that blue robot radio range ends like if there are no breadcrumbs from orange robot.

caguero, 2021-04-07:
@m3d , I've been studying your logs and the results that you're observing are mainly due to two reasons:

1. The breadcrumb location. To give you an example, see the picture below. I highlighted with yellow circles, two breadcrumbs that can't communicate each other because they're not close enough. It's right at the threshold (represented with a red line) but far enough to create a disconnected segment.

[image]

2. The fact that tunnel tiles are small compared with other types (e.g. urban). Every time that we traverse a tile, we introduce a visibility cost. Therefore, when the tiles are small, the density of breadcrumbs need to be higher.

I just created pull request #884 to mitigate (2). The coverage should be better in tunnels with the same amount of breadcrumbs, making it closer to the other segment types.

My resolution is that this was expected behavior but we just created a patch to make the behavior more consistent in tunnels.

m3d, 2021-04-07:
@caguero interesting, thank you for investigation. I am looking forward to new release to test it both in Finals and old Tunnel worlds.

closed 2021-04-07 by caguero