osrf/subt #812: Individual Motor Control (RPY + Thrust)

bfotheri, 2021-03-08:
In tunnel circuit the cmd_vel topic mapped to an RPY thrust controller for UAVs. We're trying to experiment with some trajectory planners that output in that format even if said control won't be supported for the competition.

Trouble is, the old ignition topics for individual motor control are gone. The /model/X4/command/motor_speed topic is nowhere to be seen when an "ign topic -l" is issued. The other topics are present but not the one's associated with the libignition-gazebo-multicopter-motor-model-system.so plugin. 

If this is by design is there a way we can get access to those topics again? And if it is not reproducible on your end, I will do a reinstall on my system. Thanks for you help!

nkoenig, 2021-03-19:
The `ign topic -l" will only show advertised topics. See [this issue](https://github.com/ignitionrobotics/ign-transport/issues/39). 

Have you tried connecting to the `motor_speed` topic?

bfotheri, 2021-04-01:
I believe I am lacking understanding of something else or have a bug in my setup. When I issue `ign topic -l` when the subt simulator is running it doesn't return any ignition topics when there should be at least some being advertised while the sim is running.

I need to revisit this and try a few more things so I can be more helpful on this issue. I'll try to get an update on this soon.

nkoenig, 2021-10-11:
You will need to set the `IGN_TRANSPORT_TOPIC_STATISTICS=1` environment variable.