osrf/subt #818: Simple Tunnel 2 - stuck robot under tent/base station

m3d, 2021-03-12:
I see strange situation in currently running simulation `cbc627d5-0ae0-4403-a132-4be7d4da1b00` (robotika), where the robot "accidently" explored the starting area and got stuck under the tent:
[image]
I can see wheels turning but robot is not moving ...
(I will know more when the run is over, maybe)

iche033, 2021-03-13:
turns out to be an issue with the collisions:

[image]

I've just fixed the collisions and uploaded the model onto Fuel: https://app.ignitionrobotics.org/OpenRobotics/fuel/models/Base%20Station

m3d, 2021-03-13:
Thanks - from our side it was rather "accident", but thanks for fix (I am not sure if I can easily reproduce it)

closed 2021-03-13 by iche033