osrf/subt #821: The big thermocam theft :-D

peci1, 2021-03-15:
It seems somebody stole all thermocams of the already accepted models...

E.g. MARBLE_HUSKY_SENSOR_CONFIG_3 accepted in #779 is now at version 5 at fuel:
https://fuel.ignitionrobotics.org/1.0/openrobotics/models/marble_husky_sensor_config_3/tip/files/model.sdf . There's no thermal camera there.

Version 4 has it https://fuel.ignitionrobotics.org/1.0/openrobotics/models/marble_husky_sensor_config_3/4/files/model.sdf .

I guess this might have happened because none of the 3 accepted models has the thermocam in the model.sdf submitted to this repo. What was the reason for manual modifications of the model.sdf files on fuel only? This only seems like another maintenance nightmare...

@nkoenig @adlarkin

iche033, 2021-03-16:
I've fixed the models on Fuel. Thanks for spotting the issue..

There is ongoing effort that adds better version control support on Fuel so users can get a diff when updating the models.

peci1, 2021-03-16:
Thanks, Fuel is now up-to-date and contains the thermal cameras.

closed 2021-03-16 by peci1