osrf/subt #824: Fuel model error: Tunnel Tile 6 Rails

peci1, 2021-03-16:
Model https://app.ignitionrobotics.org/OpenRobotics/fuel/models/Tunnel%20Tile%206%20Rails has broken metadata.

The metadata.pbtxt contains the following part:

```
dependencies {
  uri: "https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Tunnel Tile 6"
  uri: "https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Rail Tunnel Elevation"
}
```

And `ign fuel download` looks like:

```
$ ign fuel download -v4 --url "https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Tunnel Tile 6 Rails/0"
Downloading model: 
  Name: tunnel tile 6 rails
  Owner: openrobotics
  Server:
    URL: https://fuel.ignitionrobotics.org
    Version: 1.0

[Msg] Downloading model [https://fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/metadata.pbtxt]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/model.config]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/model.sdf]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/thumbnails/]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/thumbnails/1.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/thumbnails/2.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/thumbnails/3.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/thumbnails/4.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/thumbnails/5.jpg]
[libprotobuf ERROR google/protobuf/text_format.cc:298] Error parsing text-format ignition.msgs.FuelMetadata: 23:6: Non-repeated field "uri" is specified multiple times.
Download succeeded.
```

Looking at [the definition of the metadata message](https://github.com/ignitionrobotics/ign-msgs/blob/ign-msgs6/proto/ignition/msgs/fuel_metadata.proto), it seems to me the metadata might really be wrong as it contains only one `dependencies` section with two `url` fields.

There's also comment

https://github.com/ignitionrobotics/ign-msgs/blob/521b861bc7cfe8276a38ad58409f1b6307a88e17/proto/ignition/msgs/fuel_metadata.proto#L144

which is obviously out of date now.

iche033, 2021-03-16:
The `metadata.pbtxt` files for the `Tunnel Tile * Rails` models have been fixed on Fuel

peci1, 2021-03-16:
Thank you, I verified these models are now fixed.

Shouldn't Fuel somehow check validity of these files before allowing upload? It also seems the design of the metadata message definition is a bit cumbersome and I can imagine the way it was in Tunnel Tile 6 Rails before would be actually nicer to use and easier to understand. But I'm not sure how big a problem it would be to change the definition. Maybe keeping the `Dependencies` field repeated, but making also the `uri` field repeated, would keep backwards compatibility? (at least with the text-based messages used on Fuel).

I'm having one more fuel-related issue, though.

The dependency "Tunnel Tile 1" reports tip version number 3 on https://app.ignitionrobotics.org/OpenRobotics/fuel/models/Tunnel%20Tile%201 . However, there is a mysterious version 6 that somehow sneaked in my cache around a year ago (that would be the time I started playing around with the SubT sim). How could that happen? Did the versioning of "Tunnel Tile 1" get reset at some point?

```
peci1@cras-11:~ $ ls -l "/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1"
drwxrwxr-x   5 peci1 peci1  4096 Mar 17 00:00 3/

peci1@cras-11:~ $ ls -l "/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/Tunnel Tile 1"
drwxrwxr-x   5 peci1 peci1  4096 Mar 21  2020 6/
```

I'll resolve the issue by deleting all these outdated models, but still, shouldn't the versioning be non-decreasing?

Presence of this outdated version brings me this kind of problems when running Ignition Gazebo with a world that contains Tunnel Tile 1:

```
[Msg] Downloading model [https://fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/scripts/]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/scripts/tunnel_tile.material]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/Chainlink_Albedo.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/Chainlink_Albedo.png]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/Chainlink_Metalness.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/Chainlink_Metalness.png]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/Chainlink_Normal.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/Chainlink_Normal.png]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/Chainlink_Roughness.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/Chainlink_Roughness.png]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/ConcreteRubble_Albedo.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/ConcreteRubble_Albedo.png]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/ConcreteRubble_Metalness.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/ConcreteRubble_Metalness.png]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/ConcreteRubble_Normal.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/ConcreteRubble_Normal.png]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/ConcreteRubble_Roughness.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/ConcreteRubble_Roughness.png]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/ConstrainedWall_Albedo.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/ConstrainedWall_Normal.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/3/materials/textures/Entrance.jpg]
[... cut ...]
[Err] [SystemPaths.cc:364] URI [https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Tunnel Tile 1/tip/files/materials/textures/Light_Normal.jpg] resolved to path [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1/6/materials/textures/Light_Normal.jpg] but the path does not exist
[Err] [SystemPaths.cc:444] Could not resolve file [https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Tunnel Tile 1/tip/files/materials/textures/Light_Normal.jpg]
[Err] [SceneManager.cc:559] Unable to find file [https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Tunnel Tile 1/tip/files/materials/textures/Light_Normal.jpg]
```

iche033, 2021-03-17:
> Shouldn't Fuel somehow check validity of these files before allowing upload?

I think that could be a nice-to-have feature on fuel, i.e. a simple XML / protobuf validation check. I don't know if that should prevent the upload though - maybe an optional feature that users can turn on.

> but making also the uri field repeated

Makes sense. I don't know if there is any plan on adding other fields in `dependencies` or not. If so, it would probably be better to stay as as a repeated field.

> Did the versioning of "Tunnel Tile 1" get reset at some point?

It seems like that someone may have removed and re-added this model by accident. We should make sure that we don't do that for released models.

nkoenig, 2021-06-21:
I've created an issue to [track file validation](https://gitlab.com/ignitionrobotics/web/fuelserver/-/issues/8), and I'll close this issue.

closed 2021-06-21 by nkoenig