osrf/subt #830: TCP Bridge appears to be dying

dan-riley, 2021-03-19:
As of the last couple days, we're unable to run any cloudsim runs successfully.  All of the robots stop moving, many of them shortly after the run starts.  We made no significant changes that should affect mobility to our docker image which had successfully scored previously.  I should note this all occurs in the Finals Qual world; I haven't tried other worlds.  

I narrowed it down to what appears to some TCP connection to the bridge dying.  In Robot/log/rostopic_*.log, I'm seeing this, for example.  (in this run, 3 robots appeared to have the same sequence of errors around the same time, and the 4th robot had it later on)

[rospy.internal][ERROR] 2021-03-18 13:37:19,293: Unable to initiate TCP/IP socket to 172.17.202.79:44919 (http://172.17.202.79:40385/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 562, in connect
    self.read_header()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 657, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 600, in _validate_header
    raise TransportInitError("remote error reported: %s"%header['error'])
TransportInitError: remote error reported: node shutting down

[rospy.internal][WARNING] 2021-03-18 13:37:20,294: Unknown error initiating TCP/IP socket to 172.17.202.79:44919 (http://172.17.202.79:40385/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 560, in connect
    self.socket.connect((dest_addr, dest_port))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 111] Connection refused

Then, a few minutes later after some connections are added again:

[rospy.internal][ERROR] 2021-03-18 13:41:56,777: Unable to initiate TCP/IP socket to 172.17.202.79:38223 (http://172.17.202.79:37309/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 562, in connect
    self.read_header()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 657, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 600, in _validate_header
    raise TransportInitError("remote error reported: %s"%header['error'])
TransportInitError: remote error reported: node shutting down

[rospy.internal][WARNING] 2021-03-18 13:41:57,779: Unknown error initiating TCP/IP socket to 172.17.202.79:38223 (http://172.17.202.79:37309/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 560, in connect
    self.socket.connect((dest_addr, dest_port))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 111] Connection refused

[rospy.internal][INFO] 2021-03-18 13:41:57,779: topic[/statistics] removing connection to http://172.17.202.79:37309/
[rospy.internal][INFO] 2021-03-18 13:42:28,205: topic[/statistics] adding connection to [http://172.17.202.79:35447/], count 52
[rospy.internal][ERROR] 2021-03-18 13:42:28,646: Unable to initiate TCP/IP socket to 172.17.202.79:46227 (http://172.17.202.79:44029/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 562, in connect
    self.read_header()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 657, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 600, in _validate_header
    raise TransportInitError("remote error reported: %s"%header['error'])
TransportInitError: remote error reported: node shutting down

[rospy.internal][INFO] 2021-03-18 13:42:28,684: topic[/statistics] adding connection to [http://172.17.202.79:44413/], count 53
[rospy.internal][WARNING] 2021-03-18 13:42:29,647: Unknown error initiating TCP/IP socket to 172.17.202.79:46227 (http://172.17.202.79:44029/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 560, in connect
    self.socket.connect((dest_addr, dest_port))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 111] Connection refused

[rospy.internal][INFO] 2021-03-18 13:42:29,647: topic[/statistics] removing connection to http://172.17.202.79:44029/

Then the next thing is the normal error when the run ends and the connection is shut down.

As I mentioned, I don't think we changed anything that could cause this so it seems like it's a problem from the bridge, but I'm not 100% sure.  Thanks!

nkoenig, 2021-03-19:
Do you know the group_id of this simulation run?

dan-riley, 2021-03-19:
@nkoenig I assume you mean the id that is displayed on the submission page?  This latest one was bac91270-c228-4508-b4ac-87eda1bc7a76.

dan-riley, 2021-03-21:
Just an update, I did find an error in one of our nodes that would cause the robots to stop moving, so it could have been on our end after all.  However, I'm still curious about those TCP errors, since our node shouldn't affect those (unless our node crashing could cause it since it has publishers and subscribers?).  

On a related note, do any of the logs have the console output for errors from a node crashing?  I haven't been able to find any.

zwn, 2021-03-21:
If you are asking about console stacktrace then that is not available in the logs. Only ROS logs are saved, stdout and stderr are not saved.

dan-riley, 2021-03-21:
Yeah, that's what I thought.  Guess the only way to do is to catch every possible error (but then the error wouldn't occur if I knew it was there!).

malcolmst, 2021-03-29:
I think I remember seeing those TCP connection errors several months back when investigating something else. Can't remember all the details at the moment, but I believe they're not a cause for concern.

Also, for debugging crashes you could maybe try emitting an ROS message from a `SIGSEGV` handler (e.g. with `backtrace` to get the stack trace), but I haven't actually tried this with ROS so not 100% sure it'll work. Used something like that in another project though, storing the stack trace to a mongo database, and it came in handy :).

closed 2021-05-10 by dan-riley