osrf/subt #833: Space-rocket-like UGV in cloudsim

peci1, 2021-03-21:
> greatest_distance_traveled: 36945.09383365675

Run ID: bcbeee57-026e-4af9-8244-0626745fe

Robot X1 was driving normally until time 2480, then it skyrocketed... There's no event logged around that time. The simulation ended at time 2600 for some reason, though.

Pos-data excerpt:

```
2477 232000000 135.854 -126.486 25.1878
2478 236000000 135.404 -125.634 25.2322
2479 240000000 134.816 -124.981 25.3128
2479 900000000 134.814 -124.529 24.4071
2480 100000000 134.826 -124.095 23.4863
2480 248000000 134.785 -123.66 22.5639
2480 376000000 134.736 -123.218 21.6301
2480 488000000 134.713 -122.772 20.7321
2480 592000000 134.732 -122.287 19.8394
2480 688000000 134.796 -121.765 18.9667
2480 776000000 134.901 -121.215 18.1262
2480 860000000 135.043 -120.623 17.2887
2480 940000000 135.214 -119.997 16.461
2481 16000000 135.41 -119.351 15.6511
2481 88000000 135.623 -118.695 14.8684
2481 156000000 135.848 -118.039 14.1231
2481 224000000 136.096 -117.347 13.3805
2481 288000000 136.348 -116.663 12.6918
2481 352000000 136.614 -115.943 12.0188
2481 416000000 136.89 -115.183 11.3649
2481 480000000 137.172 -114.382 10.7309
2481 540000000 137.439 -113.594 10.153
2481 600000000 137.707 -112.77 9.58781
2481 660000000 137.978 -111.915 9.03122
2481 716000000 138.237 -111.09 8.51515
2481 772000000 138.504 -110.243 7.99828
2481 828000000 138.784 -109.381 7.47719
2481 884000000 139.083 -108.507 6.94969
2481 936000000 139.382 -107.689 6.45364
2481 988000000 139.705 -106.871 5.95252
```

nkoenig, 2021-06-24:
Is run id `bcbeee57-026e-4af9-8244-0626745fe` correct? The greatest distance traveled for this run was 5.010967310766238, and the run lasted 886seconds.

nkoenig, 2021-06-24:
Looks like there was a bad level.
[image]

closed 2021-06-30 by nkoenig