osrf/subt #837: stereo camera output error in costar_shafter_sensor_config_1

guo-zixuan, 2021-03-24:
Hello guys, My team want to test both stereo visual and 3d lidar slam in subt simulator.So I turn to costar_shafter_sensor_config_1.
But when I  change  `stereo_cam:=0` to  `stereo_cam:=1`  at `line 186` in `src/subt/submitted_models/costar_shafter_sensor_config_1/launch/spawner.rb`. ROS topic can be found.However, there is no message in stereo image.May I change anything else?

peci1, 2021-03-24:
There's no stereo camera named `camera_front_right` on shafter (looking in its model.sdf). The parameter you changed is just a copy-paste leftover from some X3 or X4 UAV and on those it enables passthrough of data from `camera_front_right` and `camera_front_left`.

guo-zixuan, 2021-03-25:
Thanks for your reply ! I will try to do something else to configure my robots.Do you have any suggestions on configuring a robot with a stereo camera and a 3D lidar in this simulator?

guo-zixuan, 2021-03-25:
Maybe I just need to follow this website [https://github.com/osrf/subt/wiki/Model%20Submission%20Process](url).I will have a try.Thanks again.

peci1, 2021-03-25:
Explorer X1 robot has a 3D lidar and 4 depth cameras for example. Or wait for tomorrow and our Marmotte robot (3D lidar and 1 realsense).

closed 2021-10-11 by nkoenig