osrf/subt #840: Error for bridge breadcrumb/deploy/remaining

m3d, 2021-03-28:
Hello,
just checking that it is expected bridge error:
```
rocess[A100R/ros_ign_bridge_breadcrumbs_remaining-39]: started with pid [809]
[ERROR] [1616785379.079247599]: Failed to create a bridge for topic [/model/A100R/breadcrumb/deploy/remaining] with ROS type [std_msgs/Int32] and Ignition Transport type [ignition.msgs.Int32]
```
 in the latest release
```
cloudsim_bridge_latest: Pulling from osrf/subt-virtual-testbed
Digest: sha256:05582634755833a73fa865c1a555df2f102f9246c5bd4f49741c6ad72624bb69
Status: Image is up to date for osrf/subt-virtual-testbed:cloudsim_bridge_latest
docker.io/osrf/subt-virtual-testbed:cloudsim_bridge_latest
```
thanks
   Martin/Robotika

m3d, 2021-03-28:
https://github.com/osrf/subt/wiki/release_notes
* Add breadcrumb remaining ROS topic
still needs a ros_ign release

... so it is expected, probably ...

iche033, 2021-03-29:
yes a new ros_ign release should resolve this issue

nkoenig, 2021-04-06:
FYI, a new `ros_ign` release has been made.

acschang, 2021-06-11:
@m3d Did the `ros_ign` release fix the reported issue?

closed 2021-10-11 by nkoenig