osrf/subt #861: Timeout when running Arifacts Verification Tool from tutorial

atestee, 2021-03-31:
Hello, I am trying to run the Artifact Verification Tool using this tutorial https://github.com/osrf/subt/wiki/Artifacts%20Validator%20Tool. I am able to launch the world without a problem, but when running these commands:

`ign service -s /artifact/scan --reqtype ignition.msgs.StringMsg --reptype ignition.msgs.StringMsg --timeout 1000 --req 'data: ""'`

 `ign service -s /artifact/next --reqtype ignition.msgs.StringMsg --reptype ignition.msgs.StringMsg --timeout 1000 --req 'data: ""'`

I get timeout and nothing happens in the simulator. I am using the docker container, but also when trying this in the catkin workspace the same thing happens. 

When listing the services using `ign service -l` there isn't `/artifact/scan` or `/artifact/next`.

atestee, 2021-03-31:
Unfortunately this doesn't make any difference.

caguero, 2021-04-01:
It seems that for some reason `IGN_TRANSPORT_TOPIC_STATISTICS` interferes with `ign`. Make sure that `IGN_TRANSPORT_TOPIC_STATISTICS` is not set to `1` when you launch the world. You can disable it in `competition.ign` setting its value to `0`.

caguero, 2021-04-01:
Could you run `env` in the same terminal that you're trying to run `ign`?

caguero, 2021-04-01:
Also, could you run `ign topic --versions`?

atestee, 2021-04-06:
`ign service --versions` returns 9.1.0

`env` returns: 

`LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
LD_LIBRARY_PATH=/home/developer/subt_ws/install/lib:/opt/ros/melodic/lib:/usr/lib/x86_64-linux-gnu:/usr/lib/i386-linux-gnu:/usr/local/nvidia/lib:/usr/local/nvidia/lib64
IGN_LAUNCH_CONFIG_PATH=/home/developer/subt_ws/install/share/subt_ign/launch
IGN_GAZEBO_RESOURCE_PATH=/home/developer/subt_ws/install/share/subt_ign/worlds:/home/developer/subt_ws/install/share/ign_migration_scripts/worlds:/home/developer/subt_ws/install/share/subt_ign/worlds:/home/developer/subt_ws/install/share/ign_migration_scripts/worlds:
DISPLAY=:1
HOSTNAME=atestee-ubuntu
ROS_ETC_DIR=/opt/ros/melodic/etc/ros
USERNAME=developer
NVIDIA_VISIBLE_DEVICES=all
QT_X11_NO_MITSHM=1
PWD=/home/developer/subt_ws
LINES=53
HOME=/home/developer
CMAKE_PREFIX_PATH=/home/developer/subt_ws/install:/opt/ros/melodic
SUBT_IMAGES_PATH=/home/developer/subt_ws/install/share/subt_ign/images
ROS_ROOT=/opt/ros/melodic/share/ros
ROS_MASTER_URI=http://localhost:11311
ROS_VERSION=1
TERM=xterm
ROS_PYTHON_VERSION=2
IGN_LAUNCH_PLUGIN_PATH=/home/developer/subt_ws/install/lib:/home/developer/subt_ws/install/lib:
NVIDIA_DRIVER_CAPABILITIES=graphics,compat32,utility
SHLVL=1
PYTHONPATH=/home/developer/subt_ws/install/lib/python2.7/dist-packages:/opt/ros/melodic/lib/python2.7/dist-packages
ROS_PACKAGE_PATH=/home/developer/subt_ws/install/share:/opt/ros/melodic/share
IGN_TRANSPORT_TOPIC_STATISTICS=1
ROSLISP_PACKAGE_DIRECTORIES=
XAUTHORITY=/tmp/.docker.xauth
PATH=/opt/ros/melodic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
PKG_CONFIG_PATH=/home/developer/subt_ws/install/lib/pkgconfig:/opt/ros/melodic/lib/pkgconfig
ROS_DISTRO=melodic
_=/usr/bin/env`

nkoenig, 2021-10-11:
Unable to reproduce this problem.