osrf/subt #876: Fog in CloudSim web visualization

m3d, 2021-04-03:
Is there a plan to visualize output of `fog emitter` in CloudSim web GUI? I was surprised what is our robot doing in the straight tunnel in `Finals Practice 1` until I have seen the logged data. And I was even more surprised when I discovered that fog is even in the `Finals Qualification` world:
https://youtu.be/iDFf2BjvjeE?t=245
... so I am just checking if it is expected and if yes, if there is a plan to implement it there too?
thank you
   Martin/Robotika Team
p.s. I have not noticed the helmet in the fog until my colleague pointed it out - nice joke :-)

nkoenig, 2021-04-06:
Yes, this is under active development. Take a look at: https://github.com/ignitionrobotics/ign-launch/pull/104

m3d, 2021-04-06:
OK, thanks for info - should I close this "issue"?

nkoenig, 2021-04-08:
I can close this when fog has been released.