osrf/subt #879: Question about updated robot costs

malcolmst, 2021-04-05:
I just noticed the X1-X4 robot costs were recently doubled in the simulator (e.g. X2 config 8 was previously 135 credits, now it is 275). Looking at the API documentation, it looks like this was updated in February, but I had missed the change as the simulator had not updated with the new costs. Although I'm always excited to see what new robots are available, with my _current_ team of robots, this reduces the number I can use from 8 to about 4, which clearly has a negative impact on exploration ability.

Looking at the change history, when the robot costs were decreased last August, this affected all the robots:
https://github.com/osrf/subt/wiki/Robots/_compare/b4fc76dbf51b5d9c47358f3869c8d89585184200...591bd82a30c565c74f6af2681d3d280cb9333309

Now, with the increased costs, this only affects X1-X4:
https://github.com/osrf/subt/wiki/Robots/_compare/7cb778f11b7fadf7e4f490ed198b76b4dd3c43da...0b1ddd2918df0b90a88426fbe1c2cec4b74b020e

I had a couple questions about this change:
1) How was this decided that the cost of these specific robots should be increased, versus the others? Is this change intended to reduce robot team size (e.g. for performance reasons)?
2) There appear to be some instances where the costs vs. capabilities don't seem to be all that comparable. For example, MARBLE_HD2 config 2, with a *lot* of sensors and 12 breadcrumbs, is 180 credits, whereas an X2 config 8, with much fewer sensors and only 6 breadcrumbs, is 275 credits. Is this intentional?

Thanks

m3d, 2021-04-05:
Hi Malcolm,
I guess that they want to discourage us to use artificial models X1 to X4, but I am also curious about official answer ...

peci1, 2021-04-05:
Here's the [relevant Discourse post](https://community.subtchallenge.com/t/new-subt-virtual-testbed-features-and-model-updates/1517) which explains it quite well, I think:

>  the valuations (in SubT Credits) of the X1, X2, X3, and X4 robot platforms have been increased to their original amounts. DARPA continues to be interested in highlighting their cost-effective Systems-based robot model counterparts, which are based on real-world platforms.

I only consider it a pity that even models like EXPLORER_X1 were affected by the cost. It seems to me that this model was prepared in order to reflect a real Husky with a real sensor pack, but team Explorer chose the X1 platform because of its similarity to the Husky robot. There's very little difference between X1 and Husky in my view...

There was some time after this cost change when you were able to submit 1000+ credit teams to cloudsim, but it has been "fixed" a few weeks ago, which is probably when you noticed this change.

Good news is that you can await lots of new robot models in April!

malcolmst, 2021-04-06:
Thanks @m3d and @peci1! Guess I must have skimmed over that part in the discord post before :). 

That reasoning does make sense, though I agree it's unfortunate about some of the models that were caught up in the change. Will be looking into all the new robot choices in great detail very soon anyway.

angelacmaio, 2021-04-07:
As mentioned by @peci1, the cost updates were announced in [this SubT Community Forum post](https://community.subtchallenge.com/t/new-subt-virtual-testbed-features-and-model-updates/1517).

closed 2021-04-07 by angelacmaio