osrf/subt #880: Simulator High VRAM usage

rsawtell, 2021-04-06:
Recent changes seem to have greatly increased the amount of VRAM needed to run the simulator. As below, running a single robot is using a combined 5.5gb of vram across 2 processes without our code running (I only ran a single instance of the simulator). Is there any way to reduce this as it doesn't leave much room for anything else running on the system?

[image]

zwn, 2021-04-06:
Don't run the gui and you will get half of the memory back (for more detailed description of my experience see https://github.com/osrf/subt/issues/680#issuecomment-715146940).

peci1, 2021-04-06:
This also depends on the world you're using and whether you use `levels` or not. It will also maybe depend on the robot and how much cameras does it have.

rsawtell, 2021-04-06:
Choice of robot didn't make much difference for me, it does appear the world choice matters significantly. I'm running in final practice 1 now and the combined simulator+gui usage is only ~2.1gb (vs the 5.5gb for final qual) with the same robot I used before. I've also noticed the final qual map will freeze for a few minutes at startup as the vram is gradually allocated, so there is something in particular about that world that is contributing to high vram usage.