osrf/subt #881: Finals Practice 2 - fail to load or display parts of the world

m3d, 2021-04-06:
I have already seen it several times so I guess that it is not network issue on my side:
[image]
[image]

peci1, 2021-04-06:
This happened to me when viewing from a computer with a limited GPU (mainly VRAM).

peci1, 2021-04-06:
https://gitlab.com/ignitionrobotics/web/gz3d/-/issues/3

m3d, 2021-04-06:
@peci1 - good points, thanks. I just tested it on different machine, different browser, 32GB RAM ... outside view and that particular parts were still not loaded/drawn:
[image]

peci1, 2021-04-06:
32 GB GPU RAM? because that's what was limiting in my case

m3d, 2021-04-06:
:) sorry, then the web view is not for normal users ... no pure RAM, some ordinary GPU/Lenovo P50/P52

peci1, 2021-04-06:
I had 2 GB shared GPU memory on the failing computer. Many people with dedicated GPUs have more (I never had a problem with a 4 GB GPU). Some kind of level-by-level loading would definitely help...

peci1, 2021-04-06:
Or LOD support, but that's only in Edifice...

peci1, 2021-04-06:
(or was it just in the [roadmap](https://ignitionrobotics.org/docs/all/roadmap))?

nkoenig, 2021-04-13:
Version 13 of the model should work: https://app.ignitionrobotics.org/OpenRobotics/fuel/models/Urban%20Stairwell%20Platform%20Centered. 

For reference, the transparency tags in the collada file were causing the problem.

closed 2021-04-26 by nkoenig