osrf/subt #885: Installation fails on Linux Mint

pauljurczak, 2021-04-08:
I'm using Linux Mint 19.3 Cinnamon, which is a derivative of Ubuntu 18.04. February release of Subt worked just fine with it. The newest release (2021-04-02), which has a different installation procedure, fails to install with these error messages:
```
paul@extra-mint:~/subt_ws$ rosdep install --from-paths src --ignore-src -r
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 2
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies (ROS distro is not set. Make sure `ROS_DISTRO` environment variable is set, or use `--rosdistro` option to specify the distro, e.g. `--rosdistro indigo`):
subt_ign: Cannot locate rosdep definition for [rostest]
coro_pam_sensor_config_1: Cannot locate rosdep definition for [xacro]
cerberus_anymal_c_ignition_gazebo_plugin_1: Cannot locate rosdep definition for [catkin]
robotika_x2_sensor_config_1: Cannot locate rosdep definition for [xacro]
cerberus_anymal_c_sensor_config_1: Cannot locate rosdep definition for [xacro]
robotika_kloubak_sensor_config_2: Cannot locate rosdep definition for [xacro]
subt_example: Cannot locate rosdep definition for [roscpp]
subt_ros: Cannot locate rosdep definition for [urdf]
robotika_freyja_sensor_config_1: Cannot locate rosdep definition for [xacro]
cerberus_anymal_c_sensor_config_2: Cannot locate rosdep definition for [roslaunch]
...
```
or this variant:
```
paul@extra-mint:~/subt_ws$ rosdep install --from-paths src --ignore-src --rosdistro melodic -r
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
subt_ign: Unsupported OS [mint]
coro_pam_sensor_config_1: Unsupported OS [mint]
cerberus_anymal_c_ignition_gazebo_plugin_1: Unsupported OS [mint]
robotika_x2_sensor_config_1: Unsupported OS [mint]
cerberus_anymal_c_sensor_config_1: Unsupported OS [mint]
robotika_kloubak_sensor_config_2: Unsupported OS [mint]
subt_example: Unsupported OS [mint]
subt_ros: Unsupported OS [mint]
robotika_freyja_sensor_config_1: Unsupported OS [mint]
cerberus_anymal_c_sensor_config_2: Unsupported OS [mint]
...
```

peci1, 2021-04-12:
What if you try with `--os=ubuntu:bionic`?

peci1, 2021-04-12:
Also, setting environment variables `OS_NAME` and `OS_VERSION` should do the same thing.

pauljurczak, 2021-04-12:
Thank you for the suggestion. I was going to try it, but my Mint 19.3 Cinnamon stopped booting after software update. I'm going to stick to Ubuntu flavors - fewer problems.

closed 2021-04-12 by pauljurczak