osrf/subt #889: X500 doesn't fly straight

rsawtell, 2021-04-13:
Easiest way to reproduce is to start teleop with it and fly straight up: it will veer to one side as it does so. Possibly not related but I also get a slew of error messages when I try to view with rviz.

`[ERROR] [1618322164.956033141, 1.792000000]: Could not load resource [package://ctu_cras_norlab_x500_sensor_config_1/meshes/sensors/realsense.dae]: Unable to open file "package://ctu_cras_norlab_x500_sensor_config_1/meshes/sensors/realsense.dae".
[ WARN] [1618322164.956484464, 1.792000000]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://ctu_cras_norlab_x500_sensor_config_1/meshes/sensors/realsense.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1618322164.957129550, 1.792000000]: Could not load model 'package://ctu_cras_norlab_x500_sensor_config_1/meshes/sensors/realsense.dae' for link 'X1/base_link': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://ctu_cras_norlab_x500_sensor_config_1/meshes/sensors/realsense.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ WARN] [1618322164.957562831, 1.792000000]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://ctu_cras_norlab_x500_sensor_config_1/meshes/sensors/realsense.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1618322164.957634545, 1.792000000]: Could not load model 'package://ctu_cras_norlab_x500_sensor_config_1/meshes/sensors/realsense.dae' for link 'X1/base_link': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://ctu_cras_norlab_x500_sensor_config_1/meshes/sensors/realsense.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1618322164.957798551, 1.792000000]: Could not load resource [package://ctu_cras_norlab_x500_sensor_config_1/meshes/sensors/bluefox.dae]: Unable to open file "package://ctu_cras_norlab_x500_sensor_config_1/meshes/sensors/bluefox.dae".
[ WARN] [1618322164.957894544, 1.792000000]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://ctu_cras_norlab_x500_sensor_config_1/meshes/sensors/bluefox.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1618322164.957953156, 1.792000000]: Could not load model 'package://ctu_cras_norlab_x500_sensor_config_1/meshes/sensors/bluefox.dae' for link 'X1/base_link': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://ctu_cras_norlab_x500_sensor_config_1/meshes/sensors/bluefox.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1618322164.958084583, 1.792000000]: Could not load resource [package://ctu_cras_norlab_x500_sensor_config_1/meshes/holders/side_mrs.dae]: Unable to open file "package://ctu_cras_norlab_x500_sensor_config_1/meshes/holders/side_mrs.dae".
[ WARN] [1618322164.958180964, 1.792000000]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://ctu_cras_norlab_x500_sensor_config_1/meshes/holders/side_mrs.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1618322164.958258481, 1.792000000]: Could not load model 'package://ctu_cras_norlab_x500_sensor_config_1/meshes/holders/side_mrs.dae' for link 'X1/base_link': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://ctu_cras_norlab_x500_sensor_config_1/meshes/holders/side_mrs.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1618322164.958388150, 1.792000000]: Could not load resource [package://ctu_cras_norlab_x500_sensor_config_1/meshes/holybro/holybro_x500_arm.dae]: Unable to open file "package://ctu_cras_norlab_x500_sensor_config_1/meshes/holybro/holybro_x500_arm.dae".
[ WARN] [1618322164.958473834, 1.792000000]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://ctu_cras_norlab_x500_sensor_config_1/meshes/holybro/holybro_x500_arm.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1618322164.958531492, 1.792000000]: Could not load model 'package://ctu_cras_norlab_x500_sensor_config_1/meshes/holybro/holybro_x500_arm.dae' for link 'X1/base_link': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://ctu_cras_norlab_x500_sensor_config_1/meshes/holybro/holybro_x500_arm.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1618322164.958640123, 1.792000000]: Could not load resource [package://ctu_cras_norlab_x500_sensor_config_1/meshes/holybro/holybro_x500_motor.dae]: Unable to open file "package://ctu_cras_norlab_x500_sensor_config_1/meshes/holybro/holybro_x500_motor.dae".
`

nkoenig, 2021-04-13:
What is your test procedure for the flying issue? I can teleoperate the X500 straight using a joystick with the following commands:

1. `ign launch competition.ign -v 4 worldName:=simple_cave_01 circuit:=cave robotName1:=X1 robotConfig1:=CTU_CRAS_NORLAB_X500_SENSOR_CONFIG_1`
2. `roslaunch subt_example teleop.launch`

rsawtell, 2021-04-13:
I just tested with your listed commands and I'm still seeing it. After launching those, I hold the right bumper down and hold up on the right joystick of my controller and while it does go up it veers off to one side as it does so.

https://user-images.githubusercontent.com/700223/114600570-81013480-9c62-11eb-8e78-1a2ee2095714.mp4

acschang, 2021-06-11:
I cannot reproduce this issue.  I issued strictly X velocities ranging of 0.25 m/s and 2.5 m/s and did not notice any veering in either case.

rsawtell, 2021-06-14:
rostopic pub -1 /X1/cmd_vel geometry_msgs/Twist -- '[0,0,150]' '[0,0,0]' should mimic what my controller appears to be outputting, and will cause the veering effect, though any sufficiently large z value will begin to do it.

Similarly, once you are in the air, any sufficiently large negative z value will cause it to do a weird downward wobble (controller seems to default to -50).

This may be a consequence of the default z values "roslaunch subt_example teleop.launch" publishes with the controller, but it does not have this effect with other UAV's so far as I have noticed.

acschang, 2021-06-14:
I am able to reproduce the issue with your posted command.

It seems the root cause of the discrepancy in performance between the X500 and other UAVs is due to the X500 using the `MRSMultirotorController` plugin and other UAVs within the repository using the `MulticopterVelocityControl` plugin.

Based upon this, I would say the quadrotor is functioning as intended and erratic behavior is to be expected with large commands.

closed 2021-06-24 by nkoenig