osrf/subt #893: Manual termination is needed after reaching maximum limit of allowed artifact reports

m3d, 2021-04-15:
According to the rules:
```
9.1.8. Run Termination
A scored run terminates upon any of the following conditions:
• Time Expiration: The given time expires before another termination criterion is met
• Disabled Platforms: All platforms are in a disabled state
• Maximum Score: All artifacts have been successfully reported and the maximum score has been achieved
• Artifact Reports Limit: The team reaches the maximum limit of allowed artifact reports
```
so the last point is the case, when the robots are stopped, but the web view is still available (it looks like this):
[image]
(note, that the score was no updated but it could be due to already stopped simulation?)

We use Teambase to limit the simulation time, but because all robots are stopped the run continues probably for the whole simulated hour? This is maybe intended so that user sees that it was terminated due to reaching the limit of allowed artifact reports? OTOH if you have set of short simulations they would be blocked.

This is not critical, so I am just curious if it is intended? It would be also nice to see it in the overview of runs what was the termination reason.

zwn, 2021-04-15:
Just a small clarification - when Artifact Reports Limit is reached, the run on cloudsim is somehow marked as "complete" but the simulation does not end.

nkoenig, 2021-04-15:
I believe the issue is resolved in https://github.com/osrf/subt/pull/894.

The `recording_complete` message was sent only once, because the `state` was not set properly. It's possible for cloudsim to miss this message.

closed 2021-04-26 by nkoenig