osrf/subt #904: Desired Feature: A sample world with lots of artifacts close to the entrance

realdealneil, 2021-04-23:
We would like to propose that a world (could be an existing world) on cloudsim be made that has lots of artifacts near the tunnel entrance so that we can do more rapid testing of artifact management between multiple agents.  This is a difficult thing to simulate locally and would be beneficial to iron out problems with team base station, etc.  The simple cave circuits would be good candidates.

nkoenig, 2021-04-26:
This is something we can work on, but I don't have a specific time-frame for you.

Alternatively, you are more than welcome to create a new world and submit it here.

realdealneil, 2021-04-27:
Very good.  We're working on it and will submit it when it's ready.

Neil