osrf/subt #905: rosservice set_rate available in docker/CloudSim?

m3d, 2021-04-25:
Hello,
I am trying to reduce frame rate for X500 drone, both front camera (~60Hz) and up/down RGBD cameras (~30Hz). According to https://github.com/osrf/subt/pull/791 I am trying to call `rosservice list | grep set_rate`  in solution docker but the output is empty.
I am using the latest docker images (2021-04-22). Is there some change needed on solution docker side?
thanks a lot
   Martin

peci1, 2021-04-25:
X500 doesn't support the ROS-side `set_rate` (yet). Each model has to explicitly add the `set_rate` ROS service bridges to support it. You can probably try to add it in a pull request, though.

peci1, 2021-04-25:
You can tell which models support it by searching `vehicle_topics.launch`. If it contains `set_rate` bridges or uses the helpers from `subt_ros/launch/models_common`, it should.

m3d, 2021-04-26:
@peci1 thanks for info

peci1, 2021-04-26:
Fixed in #906, thanks @petrlmat ;)

hilario, 2021-05-05:
Is there an approximate release date for an updated docker image with these changes?

m3d, 2021-05-18:
The `set_rate` for X500 is available in 2021-05-14 release. It works for me on CloudSim although I am still getting black or mixed images, so I believe it is some other problem on our side. I am closing this issue, thanks for help!

closed 2021-05-18 by m3d