osrf/subt #911: Problem with rosdep keys and EMESENT_HOVERMAP

pauljurczak, 2021-05-01:
I just installed v2021-04-22 (Catkin build) with a small glitch:
```
paul@desktop:~/subt_ws$ rosdep install --from-paths src --ignore-src -r
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
subt_ign: Cannot locate rosdep definition for [ignition-transport9]
subt_communication_broker: Cannot locate rosdep definition for [ignition-transport9]
bosdyn_spot: Cannot locate rosdep definition for [ignition-gazebo4]
subt_ros: Cannot locate rosdep definition for [ignition-transport9]
ctu_cras_norlab_lily_sensor_config_1: Cannot locate rosdep definition for [ignition-gazebo4]
subt_communication_model: No definition of [ignition-common3] for OS version [bionic]
ctu_cras_norlab_absolem_sensor_config_1: Cannot locate rosdep definition for [ignition-gazebo4]
subt_rf_interface: No definition of [ignition-math6] for OS version [bionic]
emesent_hovermap_sensor_config_1: Cannot locate rosdep definition for [ignition-gazebo4]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```
and it seems to run fine, except of some issues with EMESENT_HOVERMAP. It is rendered as a box:

[image]

Running the simulation produces a bunch of errors and doesn't publish Lidar data:
```
paul@desktop:~/subt_ws$ ign launch -v 4 competition.ign worldName:=tunnel_circuit_practice_01 circuit:=tunnel robotName1:=X1 robotConfig1:=EMESENT_HOVERMAP_SENSOR_CONFIG_1
.....
[Err] [SystemPaths.cc:455] File [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-drone-hipoly.dae] resolved to path [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-drone-hipoly.dae] but the path does not exist
[Err] [MeshManager.cc:172] Unable to find file[/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-drone-hipoly.dae]
[Wrn] [Physics.cc:866] Failed to load mesh from [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-drone-hipoly.dae].
[Err] [SystemPaths.cc:455] File [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-guard.dae] resolved to path [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-guard.dae] but the path does not exist
[Err] [MeshManager.cc:172] Unable to find file[/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-guard.dae]
[Wrn] [Physics.cc:866] Failed to load mesh from [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-guard.dae].
[Err] [SystemPaths.cc:455] File [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/ST-LIDAR-case-2.dae] resolved to path [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/ST-LIDAR-case-2.dae] but the path does not exist
[Err] [MeshManager.cc:172] Unable to find file[/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/ST-LIDAR-case-2.dae]
[Wrn] [Physics.cc:866] Failed to load mesh from [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/ST-LIDAR-case-2.dae].
[Err] [SystemPaths.cc:455] File [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/ST-LIDAR-hipoly.dae] resolved to path [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/ST-LIDAR-hipoly.dae] but the path does not exist
[Err] [MeshManager.cc:172] Unable to find file[/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/ST-LIDAR-hipoly.dae]
[Wrn] [Physics.cc:866] Failed to load mesh from [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/ST-LIDAR-hipoly.dae].
......
[GUI] [Err] [SystemPaths.cc:455] File [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-drone-hipoly-mount.dae] resolved to path [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-drone-hipoly-mount.dae] but the path does not exist
[GUI] [Err] [MeshManager.cc:172] Unable to find file[/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-drone-hipoly-mount.dae]
[GUI] [Err] [MeshDescriptor.cc:56] Mesh manager can't find mesh named [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-drone-hipoly-mount.dae]
[GUI] [Err] [Ogre2MeshFactory.cc:496] Cannot load null mesh [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-drone-hipoly-mount.dae]
[GUI] [Err] [Ogre2MeshFactory.cc:84] Failed to get Ogre item for [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-drone-hipoly-mount.dae]
[GUI] [Err] [SceneManager.cc:360] Failed to load geometry for visual: base_link_fixed_joint_lump__body_visual
[GUI] [Err] [SystemPaths.cc:455] File [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-guard.dae] resolved to path [/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-guard.dae] but the path does not exist
[GUI] [Err] [MeshManager.cc:172] Unable to find file[/home/paul/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/emesent_hovermap_sensor_config_1/2/meshes/navi-guard.dae]
.......
```

pauljurczak, 2021-05-01:
Additionally, description at https://github.com/osrf/subt/wiki/Robots suggests gimballed lidar and a fixed VGA camera. It seems that the camera is mounted on the pan-tilt gimbal, its resolution is not VGA, but 4096x2160 and the lidar is mounted on a rotating arm.

peci1, 2021-05-01:
The rosdep failures for ignition libraries are expected and will probably not be solved (in Melodic). 

If you need rosdep to finish without errors, you can use https://gist.github.com/peci1/797346596b54b38fd9afd4b05ab4f6c1 .

Regarding the Fuel download error, this is a bug introduced in the newest release of Ignition Fuel - https://github.com/ignitionrobotics/ign-fuel-tools/issues/183 . The fix is on its way but may take a few days until it bubbles through the buildfarm. See the referenced issue for temporary workarounds. In SubT, you can also copy the Fuel cache out of a Docker image to your host machine. That should resolve the issue.