osrf/subt #915: EMESET HOVERMAP Camera gimbal not working

rsawtell, 2021-05-14:
Trying to get the gimbal on the hovermap to move, but the twist command doesn't appear to do anything.

eg:
rostopic pub /X1/gimbal/cmd_vel geometry_msgs/Twist -- '[1000,1000,1000]' '[1000,1000,1000]'

results in no movement whatsoever.

rsawtell, 2021-05-18:
It is also incorrectly labeled as being a VGA camera when the returned image is 4k DCI (2160x4096)

PeterQFR, 2021-05-18:
Hi @rsawtell The documentation needs to catch up.. We originally output a VGA image because that is what we use in hardware,  but during the model validation exercise it was pointed out the FLIR camera's actual output is as it is set. So documentation is out of date. 
Regarding the gimbal, there is an enable topic that must be set to enable control . this only needs to be done once as per the [usage instructions ](https://github.com/osrf/subt/blob/master/submitted_models/emesent_hovermap_sensor_config_1/specifications.md#usage-instructions)