osrf/subt #942: Default branch changed?

peci1, 2021-05-28:
I see the default branch of this repo got changed from `master` to `final_event`. Which branch should we follow? The catkin install tutorial on the wiki still uses master, as well as the dockerfiles...

nkoenig, 2021-06-01:
The command `git clone https://github.com/osrf/subt` will automatically grab the default branch, which is now `final_event`. The wiki and dockerfiles don't directly specify a branch name, so they should continue to work as expected.

The `final_event` branch is similar to a release. We can target bug fixes to that branch, and target new features that will not be part of Finals to `master`.

peci1, 2021-06-01:
That's right, but e.g. some of our own scripts need a branch name, so we've put `master` there. I think this change would deserve a notification to all participants.

So is it correct to assume cloudsim images will be built from `final_event` branch?

nkoenig, 2021-06-02:
That's right, the cloudsim images are build from the `final_event` branch. I've also made a note about this change in the release notes and on the main page of the wiki.

closed 2021-06-02 by peci1