osrf/subt #943: Error downloading models

meryl-spencer, 2021-06-01:
When running $ign fuel download -v 4 -j 8 -u "https://fuel.ignitionrobotics.org/openrobotics/collections/SubT Tech Repo" 
I get the following error messages: 
[Err] [Zip.cc:161] Error opening: materials/textures/Chainlink_Albedo.jpg
[Err] [Zip.cc:161] Error opening: materials/textures/Chainlink_Transparency.jpg
[Err] [Zip.cc:161] Error opening: materials/textures/Door_Albedo.jpg
[Err] [Zip.cc:161] Error opening: materials/textures/Entrance.jpg
[Err] [Zip.cc:161] Error opening: materials/textures/GroundDirt_Albedo.jpg
[Err] [Zip.cc:161] Error opening: materials/textures/Light_Plastic_Albedo.jpg
[Err] [Zip.cc:161] Error opening: materials/textures/Light_Plastic_Emissive.jpg
[Err] [Zip.cc:161] Error opening: materials/textures/Pipes_Albedo.jpg
[Err] [Zip.cc:161] Error opening: materials/textures/Rails_Albedo.jpg
[Err] [Zip.cc:161] Error opening: materials/textures/RockWall_Albedo.jpg
[Err] [Zip.cc:161] Error opening: materials/textures/RustyMetal_Albedo.jpg
[Err] [Zip.cc:161] Error opening: materials/textures/Wood_Albedo.jpg
[Err] [Zip.cc:161] Error opening: materials/textures/YellowPlastic_Albedo.jpg
[Err] [Zip.cc:161] Error opening: meshes/
[Err] [Zip.cc:161] Error opening: meshes/Edgar_Section_01.DAE
[Err] [Zip.cc:161] Error opening: meshes/Edgar_Section_02.DAE
[Err] [Zip.cc:161] Error opening: meshes/Edgar_Section_03.DAE
[Err] [Zip.cc:161] Error opening: meshes/Edgar_Section_04.DAE
[Err] [Zip.cc:161] Error opening: meshes/Edgar_Section_05.DAE
[Err] [Zip.cc:161] Error opening: meshes/Edgar_Section_06.DAE
[Err] [Zip.cc:161] Error opening: meshes/Edgar_Section_07.DAE
[Err] [Zip.cc:161] Error opening: meshes/Edgar_Section_08.DAE
[Err] [Zip.cc:161] Error opening: meshes/Edgar_Section_09.DAE
[Err] [Zip.cc:161] Error opening: meshes/Edgar_Section_10.DAE
[Err] [Zip.cc:161] Error opening: meshes/Edgar_StartingArea.DAE
[Err] [Zip.cc:161] Error opening: model.config
[Err] [Zip.cc:161] Error opening: model.sdf
[Err] [Zip.cc:161] Error opening: thumbnails/
[Err] [Zip.cc:161] Error opening: thumbnails/1.jpg
[Err] [Zip.cc:161] Error opening: thumbnails/2.jpg
[Err] [Zip.cc:161] Error opening: thumbnails/3.jpg
[Err] [Zip.cc:161] Error opening: thumbnails/4.jpg
[Err] [Zip.cc:161] Error opening: thumbnails/5.jpg
[Err] [Zip.cc:118] Source archive does not exist: /home/maspence/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/edgar mine virtual stix/12/edgar mine virtual stix.zip
[Err] [LocalCache.cc:415] Unable to unzip [/home/maspence/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/edgar mine virtual stix/12/edgar mine virtual stix.zip]

nkoenig, 2021-06-01:
I think you might have an old/broken version of the ign-fuel-tools library. Try running the following command.

```
dpkg -l | grep ignition-fuel-tools5
```

You should see a version that is 5.1.1. If you don't then you'll likely need to run:

```
sudo apt-get update
sudo apt-get upgrade libignition-fuel-tools5-dev
```

meryl-spencer, 2021-06-01:
I believe I have the required version. 

maspence:~$ dpkg -l | grep ignition-fuel-tools5
ii  libignition-fuel-tools5:amd64     5.1.1-1~bionic     amd64        Ignition fuel-tools classes and functions for robot apps - Shared library
ii  libignition-fuel-tools5-dev:amd64     5.1.1-1~bionic     amd64        Ignition fuel-tools classes and functions for robot apps - Development files

nkoenig, 2021-06-02:
These errors are due to multiple threads unzipping a file in the same directory. The `-j 8` command line option uses 8 threads to download models. Multiple models depend on the `edgar mine virtual stix` model, which can cause these models to both download and unzip the `edgar mine virtual stix` simultaneously. 

The models files are actually there, which you should be able to confirm.

I've created the following issue to track this problem: https://github.com/ignitionrobotics/ign-fuel-tools/issues/188

acschang, 2021-06-14:
I'm glad Nate has a fix for the issue however if you're looking for a temporary work-around and removing the `-j 8` from your Ignition fuel doesn't work or you don't want to re-run the command you can try the following bash commands in your fuel directory to cleanup the empty directory structure and re-extract correctly:
```
#/bin/bash
find . -name "*.zip" | while read filename; do rm "`dirname "$filename"`/materials"; rm "`dirname "$filename"`/meshes"; rm "`dirname "$filename"`/thumbnails"; rm "`dirname "$filename"`/textures"; done;
find . -name "*.zip" | while read filename; do unzip -o -d "`dirname "$filename"`" "$filename"; done;
```