osrf/subt #954: UAV crash false positive

rsawtell, 2021-06-15:
UAV's undergoing sufficient acceleration will trigger the crash plugin, disabling their rotors despite hitting no obstacles. I've experienced this with the X500 but I assume it is not specific to any single UAV.

To replicate, use the cave environment (for sufficient vertical clearance)

rostopic pub -1 1/cmd_vel geometry_msgs/Twist -- '[0,0,8]' '[0,0,0]'
wait for the UAV to get up to speed (approximately the top of the cave entrance rock outcropping where it turns to grass)
rostopic pub -1 1/cmd_vel geometry_msgs/Twist -- '[0,0,-8]' '[0,0,0]'

To be sure, I had a second UAV at a higher elevation than this so I'm positive it was not hitting the invisible ceiling.

I suspect it can be triggered at lower velocities as it was not going this fast in the initial run where I saw it happen with our solution, but this does it for me consistently without being dependent on our solution.

rsawtell, 2021-06-16:
When I saw it happen during actual testing it was going much slower, more like 5m/s to 0 and far more horizontal than vertical, but that is harder to replicate.

I had some more time to test this morning, a little less extreme but will also trigger it:

rostopic pub -1 1/cmd_vel geometry_msgs/Twist -- '[0,0,6]' '[0,0,0]'
wait for the UAV to get up to speed (approximately the top of the cave entrance rock outcropping where it turns to grass)
rostopic pub -1 1/cmd_vel geometry_msgs/Twist -- '[0,0,0]' '[0,0,0]'

hilario, 2021-06-22:
We are also experiencing false positives with the UAV crash detector.

nkoenig, 2021-06-30:
See PR #965

iche033, 2021-07-09:
alternative fix in #973

closed 2021-07-19 by nkoenig