osrf/subt #959: events.yaml events for helmet reports missing closest artifact

fcseidl, 2021-06-21:
The ```events.yaml``` file contained in the logs for a cloud run is supposed to show when a robot reports an artifact. For example, here is the record of a successful report in a recent cloud run:

```robot: Xa
 event:
  id: 191
  type: dead_battery
  time_sec: 2179
  robot: Xe
- event:
  id: 192
  type: artifact_report_attempt
  time_sec: 2181
  reported_pose_world_frame: -8.61153 -160.719 -8.96568
  reported_pose_artifact_frame: 1.88847 -160.719 -8.96568
  reported_artifact_type: TYPE_RESCUE_RANDY
  closest_artifact_name: rescue_randy_1
  distance: 2.7189
  points_scored: 1
  total_score: 3
```

That's how the reports should be recorded in ```events.yaml```. But in the same run, I encountered this event, which does not list a closet artifact, and has a negative distance. This happens again for another helmet report in the same run.

```robot: Xg
- event:
  id: 209
  type: artifact_report_attempt
  time_sec: 3504
  reported_pose_world_frame: 10.4518 7.86749 8.89695
  reported_pose_artifact_frame: 20.9518 7.86749 8.89695
  reported_artifact_type: TYPE_HELMET
  closest_artifact_name:
  distance: -1
  points_scored: 0
  total_score: 3
```

There should always be _some_ closest artifact, even for incorrect reports, even when the closest artifact is not of the type reported. Because I've seen it for multiple helmet reports, my guess is that something specific is wrong with the way reports of helmet artifacts get handled.

peci1, 2021-06-23:
Was there at least one helmet in the world? Because I think that the closest artifact field actually denotes the closest artifact of the same type... and if there is none...

fcseidl, 2021-06-24:
Good catch @peci1! No helmets in the world, so I guess the robot just saw a helmet-shaped rock. I had a different interpretation of what that field means, but you're almost certainly right. Thanks!

closed 2021-06-24 by fcseidl