osrf/subt #961: Leaky Geometry

rsawtell, 2021-06-22:
Our old friend lidar passing through walls is still around, lidar scan of X1 (though it's not specific to any one robot) in starting area (very zoomed out):

[image]

iche033, 2021-07-06:
I'm trying to reproduce this issue. I tested with 3D long range lidar from X1 C4 and 2D lidar scan from X1 C2 on finals practice 01 world and tried moving the robot around in the staging area but I couldn't reproduce the leak. Here's what I get on rviz: 

[image]

Let me know if I'm doing something different from you.

rsawtell, 2021-07-07:
Looks like you are doing what I'm doing. Does the model update command not clear old models that it could be an older model on my system that is getting used in place of the newer ones? I updated everything this morning to be sure and it is still doing it for me.

iche033, 2021-07-08:
@mjcarroll was able to reproduce the issue and spotted some visible gaps at the vents and the entrance area when viewing at certain angle. We've patched the finals staging area mesh and uploaded a new version onto Fuel.
https://app.ignitionrobotics.org/OpenRobotics/fuel/models/Finals%20Staging%20Area

closed 2021-10-11 by nkoenig