osrf/subt #967: Collision detector is way too sensitive

fcseidl, 2021-07-02:
Robots frequently incur [collision damage](https://github.com/osrf/subt/pull/916) from minor jolts that definitely should not cause any serious damage. For instance, below is a recording of a UGV disabling itself by driving over an urban hallway threshold. The robot has been manually ordered to maintain its speed, but after it hits the thin metal strip it is too damaged to continue, demonstrating unrealistically extreme fragility. (Not to mention that even if every servo in the robot _did_ break instantly on impact, one would still expect the back wheels to clear the metal before the robot's momentum was spent.)

The github discussion linked above seems to provide the only explanation of how collision detection actually works. It suggests that every mechanical part on a robot will be completely disabled whenever the robot experiences an impulse greater than what would be produced by a 7.7 cm fall. This threshold is far too low, as any real robot would almost certainly withstand such a short fall. The result is that UGVs and UAVs alike can lose their actuators silently as a result of the tiniest impacts.

https://user-images.githubusercontent.com/53065461/124317114-a7898a00-db44-11eb-9a85-fcdec2788a01.mov

snkitche, 2021-07-07:
@nkoenig 
@angelacmaio 

Code freeze for the final event is tomorrow. We will continue to test this locally to make sure we understand the parameters and effects, but we would at least like to know if this is expected behavior. 

We are also seeing dozens of collision events in the logs with wildly varying ke, despite the most frequent types of observed behavior being similar to this. 

Please let us know if there is additional info we can provide.

angelacmaio, 2021-07-07:
Which robots have you experienced this issue with, and what are the KE difference values reported during the collision event?

fcseidl, 2021-07-07:
We've observed strange collisions with the coro allies (featured in the video above) and the ctu cras norlab x500s. The x500s are particularly bad. The event log seemingly reports the robot KE and KE threshold, but no KE difference. Below is an excerpt from one of our logfiles. The robot in question is a ctu cras norlab x500. Note that several collision events seem to occur in extremely rapid succession.

```
- event:
  id: 151
  type: collision
  ke: 6.42471
  ke_threshold: 2.54927
  link: base_link
  time_sec: 1160
  robot: Xd
- event:
  id: 152
  type: collision
  ke: 29.0043
  ke_threshold: 2.54927
  link: base_link
  time_sec: 1160
  robot: Xd
- event:
  id: 153
  type: collision
  ke: 4.98783
  ke_threshold: 2.54927
  link: base_link
  time_sec: 1160
  robot: Xd
- event:
  id: 154
  type: collision
  ke: 2.97861
  ke_threshold: 2.54927
  link: base_link
  time_sec: 1161
  robot: Xd
```

snkitche, 2021-07-07:
The ke can get much larger than that too: 
```
- event:
  id: 304
  type: collision
  ke: 315.947
  ke_threshold: 2.54927
  link: base_link
  time_sec: 1370
  robot: Xf
- event:
  id: 305
  type: collision
  ke: 117.277
  ke_threshold: 2.54927
  link: base_link
  time_sec: 1370
  robot: Xf
```
```
- event:
  id: 222
  type: collision
  ke: 3479.91
  ke_threshold: 2.54927
  link: base_link
  time_sec: 961
  robot: Xd
- event:
  id: 223
  type: collision
  ke: 9398.13
  ke_threshold: 2.54927
  link: base_link
  time_sec: 961
  robot: Xd
```
```
- event:
  id: 93
  type: collision
  ke: 240.812
  ke_threshold: 36.0002
  link: base_link
  time_sec: 233
  robot: Xa
- event:
  id: 94
  type: collision
  ke: 667.962
  ke_threshold: 36.0002
  link: base_link
  time_sec: 233
  robot: Xa
- event:
  id: 95
  type: collision
  ke: 96.6824
  ke_threshold: 36.0002
  link: base_link
  time_sec: 233
  robot: Xa
- event:
  id: 96
  type: collision
  ke: 625.672
  ke_threshold: 36.0002
  link: base_link
  time_sec: 234
  robot: Xa
- event:
  id: 97
  type: collision
  ke: 120.842
  ke_threshold: 36.0002
  link: base_link
  time_sec: 234
  robot: Xa
```

iche033, 2021-07-07:
For coro allie, I was able to reproduce the issue when going over the hallway when commanding the robot with 3-4m/s velocity cmd. I think the KE threshold can be adjusted here. What's the velocity cmd you are sending to the robot?

https://github.com/osrf/subt/pull/965 adjusted the KE values for X500 to address #954. The KE values in the logs attached are quite large though. Do you have steps to reproduce the large KE values?

fcseidl, 2021-07-07:
The allie in my video was ordered to go 3 m/s. That speed consistently resulted in fatal collisions driving over the hallway supports.

snkitche, 2021-07-07:
@iche033, regarding an adjustment of the ke threshold, I would recommend this be done by increasing the global ke_height rather than changing the individual robot thresholds. I believe the default ke_height of 0.077 is just too low for all the robots, especially since the UAVs have an additional factor >1 that further lowers the effective height.  

Increasing ke_height should increase the threshold since the threshold is 9.8*(robot_mass)*(ke_height). The comment in GameLogicPlugin.cc about this says the opposite and I have not tested this explicitly in sim to see what the actual effect is, but this should be checked. 

The scale factors defined for each robot, like the recent X500 change, are multiplicative factors for the ke difference, so inversely affect the ke threshold. If you want the UGV to use the default threshold so they have a multiplicative factor of 1 for ke difference, I'd expect the ke_height default to be set closer to 1 meter (maybe more), and then e.g. the X500 factor of 5 for ke difference means that the threshold can be interpreted as a drop from 20 cm. That still seems low, but a lot better than the current default ke_height = 0.077 and multiplicative factor of 5 giving a threshold equal to 1.54 cm.

iche033, 2021-07-07:
I tested driving over the hallway threshold with all ground robots at different velocity and the findings are:

* At 3 m/s, collisions were reported for these 3 robots
  * CORO ALLIE
  * CSIRO DTR
  * CORO ROCKY - this looks like a valid collision given that the small robot tilted upward on impact with the threshold.
* At 2 m/s, no collisions reported for all ground robots.

I also tested with a ke_height as large as 1m. However, it's now very difficult to damage ground robots, including a scenario that involves the robot falling 3-4 meters.

I can make the quick fix to lower the ke factor to fix the problem for this particular scenario (to a factor around 0.6-0.7). If we want to change ke_height, we'll need repeat and do another exhaustive set of tests for all robots. cc @acschang @angelacmaio

snkitche, 2021-07-08:
@iche033 thank you. Re: reproducing the large ke values, I don't think we can access the ke or ke difference directly through our controller, so we can't really test this locally. We are only getting the information through the event logs. We will review a few recent runs and post run ids with representative behavior.

closed 2021-07-19 by nkoenig