osrf/subt #972: Deadly trap in Finals Practice 1

m3d, 2021-07-08:
robotika, `e5ef74b7-333a-417b-a97d-fa724da16bc4` (still running), Finals Practice 1 ... there is "funny" trap catching Freyja from both sides ... is it expected?
[image]

m3d, 2021-07-08:
another view:
[image]

m3d, 2021-07-11:
There was not much what to be done from the robot's point of view ...
[image]

angelacmaio, 2021-07-14:
This is the Dynamic Collapse Obstacle, which is designed to deploy when a robot leaves the collapse region and block the path behind the robot (e.g., this robot may have destabilized the terrain in transit). The obstacle's trigger mechanism does not expressly prohibit a robot from being trapped if it reverses during the collapse or if the collapse is triggered by another robot. Your robot may be able to recognize signs of an impending collapse to avoid this case.

m3d, 2021-07-14:
Just curious - how it should behave if the robot is moving backward? (Freyja is symmetric robot)

angelacmaio, 2021-07-14:
This obstacle triggers based on a robot exiting the detection region, regardless of which direction the robot is moving. You can explore the functionality by driving a robot through the area.

closed 2021-07-14 by angelacmaio

m3d, 2021-07-14:
@angelacmaio - I uploaded video https://youtu.be/FcBhwthCQtY
 ... just to demonstrate if the robot has any chance to do anything.