osrf/subt #974: artifact proximity detector

NikolaasBender, 2021-07-10:
[Err] [SdfEntityCreator.cc:443] Could not resolve the canonical link for artifact_proximity_detector

Getting this error after pulling final event branch and trying to run any of the basic worlds.
This is the command I'm using to launch and getting the error with.
`ign launch -v 3 competition.ign circuit:=$circuit $world robotName1:=Base robotConfig1:=TEAMBASE robotName2:=X1 robotConfig2:=MARBLE_HUSKY_SENSOR_CONFIG_5 localModel:=true enableGroundTruth:=true`

I have tried removing the include for the proximity detectors from the artifacts that use it and that failed. Any suggestions or solutions are welcome.

nkoenig, 2021-07-13:
I'm unable to reproduce this issue. I tried `simple_cave_01` and `finals_practice_01` worlds. Do you have a specific world that produces this error?

NikolaasBender, 2021-07-13:
There seems to have been an issue with my clone of the subt repo. It has been fixed by a full system reset and re-cloning the subt repo. I will close this issue.

closed 2021-07-13 by NikolaasBender