osrf/subt #975: SubT Finals CloudSim release?

m3d, 2021-07-13:
Hello,
when is expected the final release of CloudSim and what are the blocking points? It is nice that there is **July 8 - September 24, 2021 code freeze for competition** but without release of dockers (last one is **2021-06-17**) it does not have the value.

I am also referring to extra comments in https://github.com/osrf/subt/issues/917#issuecomment-877724294 which is maybe "too hidden".
Thanks
   Martin/Robotika Team

```
(osgar) md@md-ThinkPad-P50:~/git/osgar$ docker pull osrf/subt-virtual-testbed:cloudsim_sim_latest
cloudsim_sim_latest: Pulling from osrf/subt-virtual-testbed
Digest: sha256:2a9fb322aa435370beab901e593a65f97caa7f2936f1c20d2e9d8a4482eb9d42
Status: Image is up to date for osrf/subt-virtual-testbed:cloudsim_sim_latest
docker.io/osrf/subt-virtual-testbed:cloudsim_sim_latest
```

angelacmaio, 2021-07-13:
The models on Fuel have been updated and docker images will be released on Docker Hub and Cloudsim shortly.

nkoenig, 2021-07-13:
The release has been made. Release notes are [here](https://github.com/osrf/subt/wiki/release_notes).

m3d, 2021-07-13:
I can confirm that there are new docker images now - download in progress. Thank you!

malcolmst, 2021-07-14:
Same here, thanks!!

closed 2021-07-14 by m3d