osrf/subt #976: Simulations are not loading

malcolmst, 2021-07-13:
While checking on a few running simulations, I am currently getting an error `Currently there are no submissions` and the simulations are not loading. I've tried reloading the webpage several times, logging out and in again, and clearing cookies.

[image]

wolfgangschwab, 2021-07-13:
I agree. We had problems starting our simulation on cloudsim: ("LaunchingNodes Error: InitializationFailed"). And now I get the same error. So there seems to be a general problem. I wrote an email to subt-help.

m3d, 2021-07-13:
The same with `robotika` - no submissions ...

realdealneil, 2021-07-13:
We have the same error.  We were running three submissions last night, and they were running fine last we checked in the evening.

nkoenig, 2021-07-13:
The service is back up. Please try and post back here if you experience any issues.

malcolmst, 2021-07-15:
FYI it appears the same issue is happening again now

meryl-spencer, 2021-07-15:
It is down again.

zwn, 2021-07-15:
Well, the deadline is approaching so maybe the stress test is uncovering yet other problems (except those that were uncovered and fixed the last time).

nkoenig, 2021-07-15:
Cloudsim seems to be working. We didn't notice an error over the past few hours. We'll continue to keep an eye on it. 

Stress is not a concern for us. Cloudsim has the capability of running hundreds of simulations simultaneously, as was done during the previous circuit events. Roughly 15 simulations have been run today (July 15th).

nkoenig, 2021-07-15:
If you do notice an outage, can you post any additional information? For example, any error message from the REST route, if you are submitting programmatically.

realdealneil, 2021-07-15:
I have been getting lots of "Error: Initialization Failed" errors when uploading through the webpage.  However, if Ioad the exact same set of robots/images through the REST interface, I'm having better luck.  So, I think the main problems have to do with the webpage way of loading things.  Also, the webpage keeps giving me this error:

Error #4002: Unauthorized request

So, I'm using the REST API exclusively now.

malcolmst, 2021-07-15:
Last night I was getting “Initialization Failed” frequently too despite using the REST API. No error was returned, and it went through all the normal steps of creating nodes and pods before failing. After retrying several times (with no changes) they started running.

malcolmst, 2021-07-16:
I’m not seeing any issues affecting the simulations once they’re running fortunately, but maybe the many users concurrently accessing the centralized part (i.e. including the web portal) could be stressing that part of the system

m3d, 2021-07-19:
Yeah, I see the problem now again:
[image]
(unfortunately there is no way how to bypass it - as far as I know - and download needed logfiles)

meryl-spencer, 2021-07-19:
When using the REST API I get the following error: 'no healthy upstream'

wolfgangschwab, 2021-07-23:
The problem is back: "Currenty there are no submissions"
And: Time out when starting a simulation via rest-service.

wolfgangschwab, 2021-07-23:
Tried again and now it works again.

m3d, 2021-07-26:
I am afraid it is back again :(

m3d, 2021-07-26:
It is back now ...

m3d, 2021-07-27:
It is broken again ... coincidentally it was at the same moment I tried to replace "Finals Preliminary Round" image, could it be related?

m3d, 2021-07-27:
... and it is working now and submission to FPR worked.

realdealneil, 2021-08-17:
The submission page is down for us right now (2021-08-17, 3:48 MT).  Can you reboot it?