osrf/subt #977: X500 Topic frequencies are excessive

realdealneil, 2021-07-13:
We have started using the CTU CRAS NORLAB X500 vehicle in some of our simulations on cloudsim.  Locally, we have seen that the X500 has several sensors that publish at high resolution and frame rate.  This makes the simulator move very slowly.  Another github issue offered a solution for fixing this locally that looked something like this:

world=finals_qual
vehname=D1
for s in up down; do IGN_TRANSPORT_TOPIC_STATISTICS=1 ign service -s \
 /world/$world/model/$vehname/link/base_link/sensor/rs_${s}/set_rate \
 --reqtype ignition.msgs.Double --reptype ignition.msgs.Empty --timeout 8000 -r 'data: 10.0'; done;
ign service -s /world/$world/model/$vehname/link/base_link/sensor/front_laser/scan/set_rate \
 --reqtype ignition.msgs.Double --reptype ignition.msgs.Empty --timeout 10000 -r 'data: 10.0'; 
ign service -s /world/$world/model/$vehname/link/base_link/sensor/camera_front/image/set_rate \
 --reqtype ignition.msgs.Double --reptype ignition.msgs.Empty --timeout 12000 -r 'data: 15.0';

Here, we issue 'ign service' calls that throttle these topics.  However, we can't really do this on cloudsim (or we don't know if this works effectively on cloudsim).  I think it would be better to decrease the output rates of these sensors across the board so that the vehicles work more effectively in the simulation.  Is this possible?

m3d, 2021-07-13:
see https://github.com/osrf/subt/issues/905 - would that `set_rate` help?

realdealneil, 2021-07-13:
@m3d, Thanks for the reference.  I was not aware that there was a ros-side service for this integrated into the different vehicle models.  I'm updating now and hopefully that will fix the problem.  I will update this thread once that's tested completely.

realdealneil, 2021-07-13:
This seems to have done the trick.  This issue can be closed.

closed 2021-07-13 by realdealneil