osrf/subt #978: Sporadically missing base_link tf frame for Marble HD2 robot

malcolmst, 2021-07-18:
I have occasionally been seeing robots not starting correctly, and have been unsure about the cause.

I just had this issue reproduce locally with the Marble HD2 robot, and decided to dig in to take a closer look. With two identical robots (one started correctly and one did not), I ran `rosrun tf view_frames` in each solution container to compare the tf graphs. It turns out the robot which did not start correctly is missing the `base_link` tf frame. 

I'll dig in a bit more if I can, as this is still running locally. I'll post an update here if I find out anything else.

[image]

malcolmst, 2021-07-18:
It appears to be the `pose_static` topic that is missing on the bad robot:

"Bad robot" solution container
```
developer@59677bb4d5b4:~/subt_ws$ rostopic echo /X2N1/pose_static
WARNING: no messages received and simulated time is active.
Is /clock being published?
```

"Good robot" solution container:
```
developer@7692b3b77ba4:~/subt_ws$ rostopic echo /X2N2/pose_static
WARNING: no messages received and simulated time is active.
Is /clock being published?
transforms: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 368
        nsecs:         0
      frame_id: "X2N2/tilt_gimbal_link"
    child_frame_id: "X2N2/tilt_gimbal_link/camera_pan_tilt"
    transform: 
      translation: 
        x: 0.02
        y: 0.0
        z: 0.047725
      rotation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
  - 
...
```

malcolmst, 2021-07-18:
Couple more points while this is still running:
- Looking at the bridge container, the corresponding ign topic _is available_ and it does reference base_link:
```pose {
  header {
    stamp {
      sec: 443
    }
    data {
      key: "frame_id"
      value: "X2N1::tilt_gimbal_link"
    }
    data {
      key: "child_frame_id"
      value: "X2N1::tilt_gimbal_link::camera_pan_tilt"
    }
  }
  name: "X2N1::tilt_gimbal_link::camera_pan_tilt"
  position {
    x: 0.02
    z: 0.047725
  }
  orientation {
    w: 1
  }
}
pose {
  header {
    stamp {
      sec: 443
    }
    data {
      key: "frame_id"
      value: "X2N1::base_link"
    }
    data {
      key: "child_frame_id"
      value: "X2N1::base_link::imu_sensor"
    }
  }
  name: "X2N1::base_link::imu_sensor"
  position {
  }
  orientation {
    w: 1
  }
}
...
pose {
  header {
    stamp {
      sec: 443
    }
    data {
      key: "frame_id"
      value: "X2N1"
    }
    data {
      key: "child_frame_id"
      value: "X2N1::base_link"
    }
  }
  name: "X2N1::base_link"
  position {
  }
  orientation {
    w: 1
  }
}
...
```

- The `ros_ign_bridge_pose_static` node is running and pingable (but is just not actively publishing):

```
developer@fae71645aca6:~/subt_ws$ rosnode info /X2N1/ros_ign_bridge_pose_static
--------------------------------------------------------------------------------
Node [/X2N1/ros_ign_bridge_pose_static]
Publications: 
 * /X2N1/pose_static [tf2_msgs/TFMessage]
 * /rosout [rosgraph_msgs/Log]

Subscriptions: 
 * /clock [rosgraph_msgs/Clock]

Services: 
 * /X2N1/ros_ign_bridge_pose_static/get_loggers
 * /X2N1/ros_ign_bridge_pose_static/set_logger_level

contacting node http://fae71645aca6:46313/ ...
Pid: 192
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound (54279 - 172.17.0.2:48428) [28]
    * transport: TCPROS
 * topic: /X2N1/pose_static
    * to: /X2N1/pose_tf_broadcaster
    * direction: outbound (54279 - 172.17.0.2:48582) [26]
    * transport: TCPROS
 * topic: /clock
    * to: /ros_ign_bridge_fae71645aca6_65_3753376215699130642 (http://fae71645aca6:39117/)
    * direction: inbound (54724 - fae71645aca6:42595) [27]
    * transport: TCPROS

developer@fae71645aca6:~/subt_ws$ rosnode ping /X2N1/ros_ign_bridge_pose_static
rosnode: node is [/X2N1/ros_ign_bridge_pose_static]
pinging /X2N1/ros_ign_bridge_pose_static with a timeout of 3.0s
xmlrpc reply from http://fae71645aca6:46313/	time=18.280983ms
xmlrpc reply from http://fae71645aca6:46313/	time=0.701904ms
xmlrpc reply from http://fae71645aca6:46313/	time=0.655890ms
```

I will need to stop this shortly since it is running in aws, but it appears that either the `ros_ign_bridge_pose_static` node is not correctly receiving the ign messages, or it is receiving them and is not publishing the ros topic for some reason.

peci1, 2021-07-18:
I see the same behavior all the time sporadically with all kinds of topics. A good example is MARV robot with the 4 flippers, which are instructed to lift up in the beginning. Sometimes, only 3 or 2 of them lift up and the rest of them is "dead" for the whole simulation. I guess this behavior has something to do with the amount of nodes being started up at once - there's probably some bug in either ign-transport or ROS... 

I was reproducing this quite reliably with 3 MARV robots - the 3rd one never moved... But I haven't tried this config for over a month, so maybe something has changed...

peci1, 2021-07-18:
I'm pretty sure there's no time to correctly solve this issue. But could we ask DARPA for a workaround? Would you check after each simulation whether each robot has moved at least a tiny bit, and if there is one that hasn't, restart the run? It'd be really a pity to lose points just because ros-ign publishers not publishing...

malcolmst, 2021-07-18:
I agree, that sounds like a good workaround. It also should be possible to detect a case like this from the rostopic stats logs, which I believe includes ros topic frequency. Maybe something could automatically detect if one of the expected topics is not being published

m3d, 2021-07-19:
We also observe situations where robot is not starting (robotika, `b0572ed1-b723-4c95-9ca7-e88d3431ccd8`, drone D). It was not receiving `air_pressure` topic - I am not sure how to check if it is identical to this reported topic? Re-run worked without any problem.

m3d, 2021-07-19:
Probably another example `9fe7d49d-f9fd-4d47-a5c8-d68272e14861` drone H (still running). Note, that part of the strategy is that robots start in different times, but this one should start immediately ...

malcolmst, 2021-07-19:
I believe I've been seeing this same issue periodically too, but haven't been taking the time to debug what is causing it until I saw it locally (well accessible to me on aws anyway). It does seem to be affect a variety of topics, actually also noticed in one run the other day a UAV was missing a depth camera topic. 

I didn't take the time to carefully debug the affected `parameter_bridge` in `gdb`, but if it helps at all I did dump the callstacks of each thread.

realdealneil, 2021-07-26:
We are seeing this same type of issue on cloudsim sometimes.  In some of the run logs from cloudsim, we have vehicles do a tf lookup and the vehicles frames don't exist at all.  For example, I launched 6 vehicles, and one of them had this issue and was unable to initialize properly.  Throughout the log file, I get the following error:

2217.564000000 WARN /A2/cartographer_node [/tmp/binarydeb/ros-melodic-cartographer-ros-1.0.0/cartographer_ros/ros_log_sink.cc:55(ScopedRosLogSink::send)] [topics: /rosout, /tf, /A2/submap_list, /A2/trajectory_node_list, /A2/landmark_poses_list, /A2/constraint_list, /A2/scan_matched_points2, /statistics] W0723 06:29:12.000000 93359 tf_bridge.cc:52] "A2" passed to lookupTransform argument target_frame does not exist. 

"A2" is the vehicle's name, so it should be able to find itself in the tf tree.  This vehicle never started moving because of this issue.

m3d, 2021-07-28:
Just info, that we are also seeing the same issue, where the transform is missing for several sim-seconds (13797cc4-056f-41de-88e3-adeca5d63a57, drone A does not have TF from time 95s to 109s). Any idea for workaround on solution side?

wolfgangschwab, 2021-08-07:
We also have problems that robots are not moving at all. In our couldsim run with Name "ver89FQ" copter A4 is not starting. It got no imu data at all. 
[image]

Is there any chance to solve this issue?