osrf/subt #979: Question about breadcrumb RSSI

malcolmst, 2021-07-18:
With pull request https://github.com/osrf/subt/pull/970 the behavior of `neighbors` changed so that nodes communicating via breadcrumbs are no longer considered neighbors. While this may make the RSSI calculations make more sense, I believe it also breaks some previous behavior that the RSSI between two communicating nodes (over a path of breadcrumbs) could be monitored to help determine whether or not a new breadcrumb is needed.

Is there any exposed method now for determining how well two nodes can communicate through a path of breadcrumbs?

The only alternatives I can currently think of are either monitoring the message drop rate as a proxy for RSSI, or ignoring RSSI altogether and just deploying breadcrumbs based on distance.

caguero, 2021-07-20:
There's no direct method for determining the quality of the communication between nodes connected via breadcrumbs. We don't want to enforce any particular algorithm but other idea besides what you suggested could be to look into ad-hoc protocols (AODV, DSR, etc.) to see how they resolve route discovery. Some of them use flooding as part of the route discovery phase, which could be equivalent to send a message to all the neighbors in the SubT case. Part of the information that could be exchanged could be the RSSI of each step. I definitely haven't tried this myself, just another idea to consider.

closed 2021-07-20 by caguero

peci1, 2021-07-21:
If you want to retain the previous behavior (i.e. all nodes able to communicate are considered neighbors), just locally revert the change made in comms client: https://github.com/osrf/subt/commit/ce5905f9c611ae19f17b0e85975b19ca5cbe654b#diff-35bbedf33bee46132aa87bf52191c09913735230eb896bd143716f6a18228924 . The RSSI will remain -Inf, though.

malcolmst, 2021-07-21:
Thanks @caguero and @peci1 for the suggestions! 

I’m keeping track of neighbors (by the old definition) in my own code, so in my case it’s not a huge deal to handle the new behavior. Exchanging RSSI as part of route discovery is a good thought, though I guess that’ll only work in cases where there is a direct LOS path to another robot (not when one robot is way out on its own and *only* using breadcrumbs to maintain comms).