osrf/subt #982: cloudsim: Solution container not launched

peci1, 2021-07-20:
Simulations fe77fd46-3588-4800-ae28-d470e0a280a9 and f85a31a7-526f-4386-9dd8-e5663b18e7f7 were launched about 15 seconds apart, both utilizing the same solution docker images for all robots and the same team of robots. They only differ in the world.

In f85a31a7-526f-4386-9dd8-e5663b18e7f7, robot X1's solution container has not even been started, wheras in the other simulation is started correctly. You can see that in the buggy simulation, X1 hasn't ever moved. Inspecting master.log, I see not even our ROS params being uploaded to the parameter server (so it is not beacuse of a runtime crash of one of our nodes). The possible reasons could be:

1. The solution container has not been started at all.
2. It has been started but roslaunch found a syntax/logic error in the launched file and stopped with an error.

I think option 2. should be ruled out because there is the other simulation using the same code which ran correctly.

Could you please investigate what happened with X1 in f85a31a7-526f-4386-9dd8-e5663b18e7f7?

m3d, 2021-07-21:
I am not sure if it is related ... I discovered new feature "public" simulation:
https://subtchallenge.world/watch
and you can see there test where X4 drone did not start
https://subtchallenge.world/submission/688f3032-31d4-44e4-8f77-8962b8062911

realdealneil, 2021-07-22:
I am now unable to get any runs to work on cloudsim.  I run a set of vehicles and images that worked yesterday and it fails to initialize now.  I have been exclusively using the REST method for uploading to cloudsim so that I have a traceable record.  

The ones that worked yesterday  (team Collembola) are:
Name: RESTfinalsv2dot9finalsqualr1 ID: 3099696d-e60e-44a5-a152-a7f9dd1426d1
Name: RESTfinalsv2dot9finalsp1r1 ID: 21f68f69-5ef5-4199-bf24-9286d9f9f5e2
(These two were the same images, just uploaded to two different worlds)

Everything I've tried uploading today (using the same script, and just changing the names of images to new valid images) have failed to initialize.  I then tried uploading the same images as the ones above.  Those are here:
Name: RESTfinalsv2dot9finalsp2rtest ID: 8ad064e7-3c8b-4e66-9c60-1ed4d5
Name: RESTfinalsv2dot9finalsqualrtest ID: af5820bb-39c7-4431-a8b0-e99893c27fe4

The first one has failed to initialize, and I'm running it again to see if it keeps failing.  Anyway, everything seems to fail to initialize now, and I don't know where to go from here.

realdealneil, 2021-07-23:
It appears that the REST interface is broken.  Is there a new parameter that has to be included (public vs private maybe?)?  I have tried to upload through REST about 12 times and they've all failed to initialize.  I uploaded the same images through the web interface, and they appear to work.

peci1, 2021-07-23:
I've started 2 simulations yesterday (after adding the `public` parameter to the REST interface) and the simulations are running (I did not change our deploy command in any way).

realdealneil, 2021-07-23:
Ok, I do see that the documentation on the wiki is now updated for the REST API and that there's a new parameter called "private" that has to be added.  I will try using this as soon as my web-uploads complete.  I do believe there's a typo in the wiki because I think the REST API expects:

-F private=true

instead of

-F private="PRIVATE"

And, it appears that that option is NOT optional...if you don't include it, your upload will fail.

peci1, 2021-07-23:
> -F private=true
> 
> instead of
> 
> -F private="PRIVATE"

I think that's just a misunderstanding - the wiki page uses the uppercase placeholders also for other "variables". The "variables" are explained below along with their possible values.

realdealneil, 2021-07-23:
@peci1, you're right, I didn't read the full documentation on the wiki.  I guess the main takeaway here is that the "private" parameter in the REST API is required, even though the wiki says it's optional.  I have now been successful in uploading some runs with the private parameter, and they are running.

nkoenig, 2021-07-26:
The `private` parameter is not required. If the parameter does not exist, then the simulation is private.