osrf/subt #983: Map decimation error?

peci1, 2021-07-22:
I had a local simulation running which was sending data to the `/cloud` topic. I saw this in the bridge container:

    [pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.

I assume it is a warning from the map decimator.

I have no more details, though - mostly, I don't have any message from the `/cloud` topic as I came to an already stopped simulation.

mjcarroll, 2021-07-23:
That does look like it would be from the map decimation.  I believe that the leaf size is currently set to 0.1 meters: https://github.com/osrf/subt/blob/ce5905f9c611ae19f17b0e85975b19ca5cbe654b/subt_ros/src/cloud_downsample.cpp#L99

So it would take a large cloud in physical dimensions (not number of points) to trigger this warning.  I think that would mean that you have a point outside INT_MAX * 0.1 or INT_MIN * 0.1.  Is there any chance there you had some erroneous data in there?

peci1, 2021-07-23:
I don't say it's impossible, but the map is generally good and I've never noticed this wrong data. However, if it's a point faaaar away, it's a question whether it would ever been seen in rviz...

mjcarroll, 2021-07-23:
> However, if it's a point faaaar away, it's a question whether it would ever been seen in rviz...

Maybe we could just do a pre-filter for the INT_MAX * 0.1 and INT_MIN * 0.1 to guarantee that there aren't any points at millions of meters away.

peci1, 2021-07-23:
sounds good

peci1, 2021-08-23:
It happened again. Here are properties of the published cloud.

Num points: 642918
X range: -53 .. 381
Y range: -202 .. 27
Z range: -27 .. 29

peci1, 2021-08-23:
The cloud also has 19 channels (x,y,z included; point step 124). Total data length is 79721832 bytes.