osrf/subt #984: docker image subt_shell does not build

wolfgangschwab, 2021-07-23:
During build of the docker image subt_shell I got the errror: 

```
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "control_toolbox"
```

I solved it by adding 

```
# package control-toolbox is needed
RUN sudo apt-get update -qq \
 && sudo apt-get install -y -qq \
    ros-melodic-control-toolbox
```
to the Dockerfile. 
Could you please add it there too?

peci1, 2021-07-23:
Weird, this package used to be installed automatically by the rosdep install step... Did some package remove the dependency from package.xml?

wolfgangschwab, 2021-07-23:
I didn't have a deeper look into it. First I run the dockerfile from master branch, which did not work. Then I run the dockerfile from finals_event branch, which also did not work. Then I changed it, and the problem was solved. ;-)

meryl-spencer, 2021-08-05:
I have the same issue

nkoenig, 2021-10-11:
See PR #1021.

closed 2021-10-11 by nkoenig