osrf/subt #985: Unable to submit to prelim finals circuit

realdealneil, 2021-07-24:
I've tried to submit a solution to the finals prelim circuit (assuming that we can replace the submission later as we've been able to in the past).  Both times, it gave me an error: "Error #5514: Circuit rule not found."  Are we not able to submit to the prelim circuit yet?

m3d, 2021-07-24:
I am not sure about the official name - I guessed `Finals Preliminary Round` but I also got
```
{
    "errcode": 5514,
    "msg": "Circuit rule not found.",
    "extra": null,
    "errid": "eb3492a7-b1a2-4cc2-a893-2f126837fefb",
    "route": "POST /1.0/simulations",
    "user-agent": "",
    "remote-address": "89.24.34.90"
}
```

m3d, 2021-07-24:
p.s. I also tried manual submission and it failed as @realdealneil described:
[image]

m3d, 2021-07-26:
this problem still persists:
```
{
    "errcode": 5514,
    "msg": "Circuit rule not found.",
    "extra": null,
    "errid": "9e338135-624c-48c3-bc8a-4c31b0d6bb16",
    "route": "POST /1.0/simulations",
    "user-agent": "",
    "remote-address": "89.24.34.90"
}
```

hilario, 2021-07-26:
I can confirm the same problem, maybe the submission is not open yet?

nkoenig, 2021-07-26:
The submission should now be live.

m3d, 2021-07-26:
I can confirm that it worked now, thanks.

m3d, 2021-07-26:
p.s. is the time somehow limiting?
```
    "created_at": "2021-07-26T17:13:01.372845045Z",
    "updated_at": "2021-07-26T17:13:01.468182231Z",
    "valid_for": "96h0m0s",
...
```

nkoenig, 2021-07-26:
The `valid_for` time is a bit confusing. It's the time that the simulation can run for, in real-time, before being terminated. It does not affect your submission to the prelims.

realdealneil, 2021-07-26:
I can confirm it's working as well.  Thanks for the update.  Closing.

closed 2021-07-26 by realdealneil