osrf/subt #986: How do we recognize our map sender works correctly?

peci1, 2021-07-26:
We've implemented sending maps to mapping server. I can see locally messages published to /cloud ROS topic in each robot solution container. I also see /cloud_throttled and /cloud_throttled_downsampled. I can view all of them in an rviz with fixed frame `darpa`.

But I never saw any map in cloudsim when I enable the "robot map" visualization.

I'm a bit confused whether the frame is correctly `darpa` or `DARPA`, but our UAV colleagues send the uppercase frame and also never saw anything...

acschang, 2021-07-26:
The cloud submitted is assumed to be in the `darpa` frame however the frame name in the `PointCloud2` message does not matter as it is [forced](https://github.com/osrf/subt/blob/ce5905f9c611ae19f17b0e85975b19ca5cbe654b/subt_ros/src/cloud_downsample.cpp#L92) to `artifact_origin` when it is republished on `/cloud_throttled_downsampled`.

I just rechecked the test submission and I can see the cumulative point cloud in `robot map`.  Could you please provide more details on the issue or a run-id to check.

peci1, 2021-07-27:
Not sure what changed on our side, but now I could see the map in cloudsim web visualization.

closed 2021-07-27 by peci1