osrf/subt #987: Describe limitations of markers sent to the mapping server

peci1, 2021-07-26:
I know that Ignition GUI (and probably also its web version) doesn't support all types of markers (e.g. TEXT type). Could you please explicitly list which marker types are supported?

nkoenig, 2021-07-26:
I think you are specifically referring to the visualization during live runs. If that is correct, then I've updated the table in the wiki to indicate visualization capabilities per topic. https://github.com/osrf/subt/wiki/API#mapping-server

peci1, 2021-07-26:
Thanks, that looks good. So if markers should not display in Ignition at all, maybe you could also add another column "RViz" with all checkmarks... Just to put there a hint that Rviz is the tool that should be used for displaying all the information.

acschang, 2021-07-26:
I've added the RViz column into the table for the Mapping Server API.  Essentially the markers are any sort of visualizations that add context to the map data sent to the mapping server that fits within the [RViz marker API](http://wiki.ros.org/rviz/DisplayTypes/Marker).

closed 2021-07-26 by peci1