osrf/subt #990: circuit was not defined, exiting now

TurtleZhong, 2021-08-07:
```bash
./subt/docker/run.bash osrf/subt-virtual-testbed:latest \             255 ↵
   cave_circuit.ign \
   worldName:=simple_cave_01 \
   robotName1:=X1 \
   robotConfig1:=X1_SENSOR_CONFIG_1

circuit was not defined, exiting now
Failed to parse /home/developer/subt_ws/install/share/subt_ign/launch/cave_circuit.ign because of circuit was not defined, exiting now
```

I follow this tutorial to launch the subt simulator, but I got this error. What shoud I do to solve it, I try 
```bash
./subt/docker/run.bash osrf/subt-virtual-testbed:latest \
```

but I got nothing, the container do not launch. What should I do?

wolfgangschwab, 2021-08-07:
You need to add the the circuit parameter. As you what to run a cave world, it should work if you add  `circuit:=cave`

TurtleZhong, 2021-08-08:
@wolfgangschwab Thanks, it works.

closed 2021-08-08 by TurtleZhong