osrf/subt #1000: AWS prelim worlds

meryl-spencer, 2021-08-18:
Can we get the prelim worlds available for testing on AWS.

meryl-spencer, 2021-08-18:
It looks like the worlds are up now but I get the following error when using the rest api
{"errcode":3013,"msg":"Invalid value in field.","extra":["Circuit:Finals Preliminary Round World 3"],"errid":"c6f5a613-3f71-47c4-9de0-b1fef99a7914","route":"POST /1.0/simulations","user-agent":"curl/7.58.0","remote-address":"35.63.66.108"}

meryl-spencer, 2021-08-18:
The error also appears when using the web interface

m3d, 2021-08-18:
ACK, I see the same
```
"errcode": 3013,
    "msg": "Invalid value in field.",
    "extra": [
        "Circuit:Finals Preliminary Round World 1"
    ],
```

nkoenig, 2021-08-19:
Can you please try again. The worlds should be available for practice.

m3d, 2021-08-19:
It works for me, thanks! (`99ecb8fc-f420-42ea-8043-b8a3c4072d37`)

closed 2021-08-24 by nkoenig