osrf/subt #1001: Very incorrect ranges from lidar

jisa, 2021-08-20:
CORO_PAM_SENSOR_CONFIG_1, about 2.5 meters above ground (see pictures below), repeatedly, i.e. not just due to noise, reports distance to ground over 10 meters. Up to this moment of complete failure, it has also been overestimating the distance by about 20 % in comparison to estimates based on other sensors. 

```
# rostopic echo /robot/bottom_scan
header:
  seq: 2325
  stamp:
    secs: 46
    nsecs: 540000000
  frame_id: "robot/base_link/bottom_laser"
angle_min: 3.14159274101
angle_max: 3.14159274101
angle_increment: nan
time_increment: 0.0
scan_time: 0.0
range_min: 0.0500000007451
range_max: 40.0
ranges: [10.705933570861816]
intensities: [0.0]
---
header: 
  seq: 2326
  stamp: 
    secs: 46
    nsecs: 560000000
  frame_id: "robot/base_link/bottom_laser"
angle_min: 3.14159274101
angle_max: 3.14159274101
angle_increment: nan
time_increment: 0.0
scan_time: 0.0
range_min: 0.0500000007451
range_max: 40.0
ranges: [10.632636070251465]
intensities: [0.0]
---
header: 
  seq: 2327
  stamp: 
    secs: 46
    nsecs: 580000000
  frame_id: "robot/base_link/bottom_laser"
angle_min: 3.14159274101
angle_max: 3.14159274101
angle_increment: nan
time_increment: 0.0
scan_time: 0.0
range_min: 0.0500000007451
range_max: 40.0
ranges: [10.578147888183594]
intensities: [0.0]
---
header: 
  seq: 2328
  stamp: 
    secs: 46
    nsecs: 600000000
  frame_id: "robot/base_link/bottom_laser"
angle_min: 3.14159274101
angle_max: 3.14159274101
angle_increment: nan
time_increment: 0.0
scan_time: 0.0
range_min: 0.0500000007451
range_max: 40.0
ranges: [10.461398124694824]
intensities: [0.0]
```
[image]
[image]

jisa, 2021-08-21:
After that, when it gets over the shaft, it underestimates distance to the bottom.

```
# rostopic echo /robot/bottom_scan
header: 
  seq: 2596
  stamp: 
    secs: 51
    nsecs: 960000000
  frame_id: "robot/base_link/bottom_laser"
angle_min: 3.14159274101
angle_max: 3.14159274101
angle_increment: nan
time_increment: 0.0
scan_time: 0.0
range_min: 0.0500000007451
range_max: 40.0
ranges: [5.705638408660889]
intensities: [0.0]
---
```

At that point, the drone is sufficiently horizontal not to be pointing the lidar to a wall:
```
# rostopic echo /robot/imu/data
header:
  seq: 12747
  stamp:
    secs: 51
    nsecs: 948000000
  frame_id: "robot/base_link/imu_sensor"
orientation:
  x: -0.0141953380262
  y: -0.0169637710734
  z: 0.600932680144
  w: 0.798993514858
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity:
  x: 0.00325
  y: 0.0285
  z: 0.87225
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration:
  x: 0.015
  y: 0.06
  z: 9.675
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
```

[image]
[image]
[image]

jisa, 2021-08-21:
I see the same effects also with SSCI_X4_SENSOR_CONFIG_2. This issues are not specific to one type of drone.

jisa, 2021-08-23:
Thanks for discovering the root cause, @iche033!

Is there going to be some one last release with the fix before the competition, or will it only become available later?

iche033, 2021-08-23:
> Is there going to be some one last release with the fix before the competition, or will it only become available later?

I don't know if there will be another release before the competition actually. That's mainly up to the organizers.

As a workaround when running locally without the fix, you can tweak the min/max angle params in the sdf, e.g. by editing the model.sdf file in your fuel cache: `~/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/coro_pam_sensor_config_1/1/model.sdf`

```
                            <min_angle>3.141492654</min_angle>
                            <max_angle>3.141592654</max_angle>
```

jisa, 2021-08-24:
I can confirm that if I increase `max_angle` in the `model.sdf` file in a local docker image, `catkin_make install` and run the simulator with `localModel:=1`, the bottom lidar starts providing meaningful values. It stops overestimating (the 20 % issue) and it reports reasonable values both before and after entering the shaft.

jisa, 2021-08-29:
Thanks for confirming.

m3d, 2021-09-27:
I guess that the "drones graveyard" is the consequence (just a wild guess)
[image]
https://youtu.be/EAPSm7udG3Q?t=4510