osrf/subt #1007: Prelim World 1 Violates Artifact Specification For Gas

rsawtell, 2021-08-24:
The SubT_Finals_Artifacts_Specification.pdf specifies "The center point of the entry door threshold (at floor level) of the room where the gas is detected. In rooms with multiple doorways, only one door will be open (i.e., any others 
will be closed or blocked) and will serve as the relevant localization point."

Prelim world 1 violates this as it has a gas artifact in a multi-door room in which both entryways are unblocked.

[image]

While this doesn't really matter for this specific world at this late hour, it might be worth reviewing gas artifacts in whatever worlds are planned for the final competition to ensure they are in accordance with the specifications.

m3d, 2021-08-24:
ACK, but this should be already fixed in https://github.com/osrf/subt/pull/998
(gas replaced by Cube)

angelacmaio, 2021-08-25:
This was resolved in #998 and all worlds should adhere to the [Artifact Specification Guide](https://subtchallenge.com/resources/SubT_Finals_Artifacts_Specification.pdf).

closed 2021-08-25 by angelacmaio

m3d, 2021-08-29:
I have just seen the latest run in CloudSim and there are false reports:
```
4:44:32.908203 received: ['TYPE_GAS', [118481, -68917, -82], 'A0W2C2400RH3600W', True]
5:00:04.120556 received: ['TYPE_GAS', [126278, -74399, -109], 'A0W2C2400RH3600W', False]
5:13:26.994236 received: ['TYPE_GAS', [135330, -74490, -126], 'A0W2C2400RH3600W', False]
```
which is kind of suspicious and close to the original report (i.e. it was maybe fixed not not updated on CloudSim?)