osrf/subt #1044: Error when using ctu_cras_norlab_spot_sensor_config_1

lichenchen1205, 2021-11-30:
I followed the instructions and executed
`LC_ALL=C ign launch -v 4 ~/subt_ws/src/subt/submitted_models/ctu_cras_norlab_spot_sensor_config_1/launch/example.ign robotName:=CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1 ros:=true champ:=true`
but 
`Dbg] [SimulationRunner.cc:527] Creating postupdate worker thread (5)
[Dbg] [SimulationRunner.cc:527] Creating postupdate worker thread (6)
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
process[bridge_logger-3]: started with pid [20573]
[ERROR] [1638274525.085019674]: Could not find parameter robot_description on parameter server
[ERROR] [1638274525.085768259]: Failed to parse urdf file
[ INFO] [1638274525.085784514]: Successfully parsed urdf file
process[CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/contact_sensor-3]: started with pid [20656]
[GUI] [Dbg] [RenderUtil.cc:1632] Create scene [scene]
Warning [Model.cc:212] Non-unique names detected in XML children of model with name[CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1].
[Dbg] [EntityComponentManager.cc:765] Using components of type [273039548655896271] / [ign_gazebo_components.Sensor].
[Dbg] [EntityComponentManager.cc:765] Using components of type [14624144930658477562] / [ign_gazebo_components.Camera].
[Dbg] [EntityComponentManager.cc:765] Using components of type [17771214158902094494] / [ign_gazebo_components.Imu].
[Dbg] [EntityComponentManager.cc:765] Using components of type [10461520141335422054] / [ign_gazebo_components.WorldPose].
[Dbg] [EntityComponentManager.cc:765] Using components of type [9810498410555695050] / [ign_gazebo_components.AngularVelocity].
[Dbg] [EntityComponentManager.cc:765] Using components of type [17373107141547024696] / [ign_gazebo_components.LinearAcceleration].
[Dbg] [EntityComponentManager.cc:765] Using components of type [1301926277928136113] / [ign_gazebo_components.GpuLidar].
[Dbg] [EntityComponentManager.cc:765] Using components of type [9853217982010720764] / [ign_gazebo_components.Material].
[Dbg] [EntityComponentManager.cc:765] Using components of type [6437904990063664725] / [ign_gazebo_components.DepthCamera].
[Dbg] [EntityComponentManager.cc:765] Using components of type [9386381083157376121] / [ign_gazebo_components.ContactSensor].
process[rostopic_stats_logger-4]: started with pid [20676]
[Dbg] [EntityComponentManager.cc:765] Using components of type [360894313363162387] / [ign_gazebo_components.Joint].
[Dbg] [EntityComponentManager.cc:765] Using components of type [4514298940308304267] / [ign_gazebo_components.JointType].
[Dbg] [EntityComponentManager.cc:765] Using components of type [16752981946841796574] / [ign_gazebo_components.JointAxis].
[Dbg] [EntityComponentManager.cc:765] Using components of type [6791213906290467105] / [ign_gazebo_components.ThreadPitch].
[Dbg] [EntityComponentManager.cc:765] Using components of type [14863055383328686534] / [ign_gazebo_components.ParentLinkName].
[Dbg] [EntityComponentManager.cc:765] Using components of type [3888602368311486678] / [ign_gazebo_components.ChildLinkName].
[Dbg] [SimulationRunner.cc:866] Loaded system [ignition::gazebo::systems::PosePublisher] for entity [201]
[Dbg] [SimulationRunner.cc:866] Loaded system [ignition::gazebo::systems::PosePublisher] for entity [201]
[Wrn] [GasEmitterDetectorPlugin.cc:217] Max concentration is lower than min concentration. Setting max to unbounded.
[Dbg] [SimulationRunner.cc:866] Loaded system [subt::GasDetector] for entity [201]
[Dbg] [EntityComponentManager.cc:765] Using components of type [9436740692853731264] / [ign_gazebo_components.JointPosition].
[Dbg] [EntityComponentManager.cc:765] Using components of type [4068291309002341380] / [ign_gazebo_components.JointVelocity].
[Dbg] [EntityComponentManager.cc:765] Using components of type [16434675394613771646] / [ign_gazebo_components.JointForce].
[Dbg] [SimulationRunner.cc:866] Loaded system [ignition::gazebo::systems::JointStatePublisher] for entity [201]
[Msg] ApplyJointForce subscribing to Double messages on [/model/CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/joint/front_left_hip_x/cmd_force]
[Dbg] [SimulationRunner.cc:866] Loaded system [ignition::gazebo::systems::ApplyJointForce] for entity [201]
[Dbg] [configurable_joint_position_controller.cpp:271] [ConfigurableJointPositionController] system parameters:
[Dbg] [configurable_joint_position_controller.cpp:272] p_gain: [250]
[Dbg] [configurable_joint_position_controller.cpp:273] i_gain: [0.2]
[Dbg] [configurable_joint_position_controller.cpp:274] d_gain: [5]
[Dbg] [configurable_joint_position_controller.cpp:275] i_max: [1]
[Dbg] [configurable_joint_position_controller.cpp:276] i_min: [-1]
[Dbg] [configurable_joint_position_controller.cpp:277] cmd_max: [1000]
[Dbg] [configurable_joint_position_controller.cpp:278] cmd_min: [-1000]
[Dbg] [configurable_joint_position_controller.cpp:279] cmd_offset: [0]
[Dbg] [configurable_joint_position_controller.cpp:283] Initial target: [initial position]
[Dbg] [configurable_joint_position_controller.cpp:286] Topic: [/model/CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/joint/front_left_hip_x/0/cmd_pos]
[Dbg] [configurable_joint_position_controller.cpp:287] PID config service: [/model/CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/joint/front_left_hip_x/0/set_pos_pid]
[Dbg] [SimulationRunner.cc:866] Loaded system [subt::ConfigurableJointPositionController] for entity [201]
[Msg] ApplyJointForce subscribing to Double messages on [/model/CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/joint/front_left_hip_y/cmd_force]
[Dbg] [SimulationRunner.cc:866] Loaded system [ignition::gazebo::systems::ApplyJointForce] for entity [201]
[Dbg] [configurable_joint_position_controller.cpp:271] [ConfigurableJointPositionController] system parameters:
[Dbg] [configurable_joint_position_controller.cpp:272] p_gain: [250]
[Dbg] [configurable_joint_position_controller.cpp:273] i_gain: [0.2]
[Dbg] [configurable_joint_position_controller.cpp:274] d_gain: [5]
[Dbg] [configurable_joint_position_controller.cpp:275] i_max: [1]
[Dbg] [configurable_joint_position_controller.cpp:276] i_min: [-1]
[Dbg] [configurable_joint_position_controller.cpp:277] cmd_max: [1000]
[Dbg] [configurable_joint_position_controller.cpp:278] cmd_min: [-1000]
[Dbg] [configurable_joint_position_controller.cpp:279] cmd_offset: [0]
[Dbg] [configurable_joint_position_controller.cpp:283] Initial target: [initial position]
[Dbg] [configurable_joint_position_controller.cpp:286] Topic: [/model/CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/joint/front_left_hip_y/0/cmd_pos]
[Dbg] [configurable_joint_position_controller.cpp:287] PID config service: [/model/CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/joint/front_left_hip_y/0/set_pos_pid]
[Dbg] [SimulationRunner.cc:866] Loaded system [subt::ConfigurableJointPositionController] for entity [201]
[Msg] ApplyJointForce subscribing to Double messages on [/model/CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/joint/front_left_knee/cmd_force]
[Dbg] [SimulationRunner.cc:866] Loaded system [ignition::gazebo::systems::ApplyJointForce] for entity [201]
[... 188 lines cut ...]
[Msg] Contact system publishing on world/example/model/CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/link/rear_right_lower_leg/sensor/contact/contact
[ERROR] [1638274525.284110052]: Could not find parameter robot_description on parameter server
[ERROR] [1638274525.285210200]: Failed to parse urdf file
[ INFO] [1638274525.285463883]: Successfully parsed urdf file
process[CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/root_pub-5]: started with pid [20799]
process[CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/champ_controller-2]: started with pid [20842]
[ERROR] [1638274525.436899600]: Could not find parameter robot_description on parameter server
[ERROR] [1638274525.437504460]: Failed to parse urdf file
[ INFO] [1638274525.437523814]: Successfully parsed urdf file
process[CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/base_to_footprint_ekf-6]: started with pid [20888]
[CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/state_estimator-4] process has died [pid 20723, exit code -11, cmd /home/developer/subt_ws/install/lib/champ_base/state_estimation_node joint_states:=joint_state __name:=state_estimator __log:=/home/developer/.ros/log/9df1bc1c-51d0-11ec-ade3-2cf05dd6d644/CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1-state_estimator-4.log].
log file: /home/developer/.ros/log/9df1bc1c-51d0-11ec-ade3-2cf05dd6d644/CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1-state_estimator-4*.log
[CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/state_estimator-4] restarting process
process[CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/footprint_to_odom_ekf-7]: started with pid [20953]
[... cut ...]
[CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1/champ_controller-2] process has died [pid 20842, exit code -11, cmd /home/developer/subt_ws/install/lib/champ_base/quadruped_controller_node __name:=champ_controller __log:=/home/developer/.ros/log/9df1bc1c-51d0-11ec-ade3-2cf05dd6d644/CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1-champ_controller-2.log].
log file: /home/developer/.ros/log/9df1bc1c-51d0-11ec-ade3-2cf05dd6d644/CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1-champ_controller-2*.log
`
I tested it on the computer and in docker and got the same result. How can I solve this problem?

peci1, 2021-11-30:
You're right, that is an error in the models. I sent a fix for this to the `master` branch (#1045). With this fix, I can launch the command you mentioned correctly.

lichenchen1205, 2021-12-01:
> You're right, that is an error in the models. I sent a fix for this to the `master` branch (#1045). With this fix, I can launch the command you mentioned correctly.

lichenchen1205, 2021-12-01:
[image]
Thank you very much for your help. I can run this command according to your method, **but when I start it, the robot will always lie there, as shown in the picture.and I use`roslaunch champ_teleop teleop.launch`，but I can’t use the keyboard to move it**.

lichenchen1205, 2021-12-22:
Another question, how do I change the world the robot is in? I replaced the example.sdf file in subt_ws/src/subt/submitted_models/bosdyn_spot/worlds, but it always fails, the loading environment appears but the robot loads The problem of failure, or the robot is not loaded when the environment is loaded, is there a correct way?
@peci1

lichenchen1205, 2021-12-22:
Another question, how do I change the world the spot is in? I replaced the example.sdf file in subt_ws/src/subt/submitted_models/bosdyn_spot/worlds, it will appear to load the environment but the robot fails to load or both failed， is there a correct way?

lichenchen1205, 2021-12-22:
Another question, how do I change the world the spot is in? I replaced the example.sdf file in subt_ws/src/subt/submitted_models/bosdyn_spot/worlds, it will appear to load the environment but the robot fails to load or both failed， is there a correct way?
@peci1

peci1, 2021-12-22:
Hi, if you followed the official install instructions, you should edit the files in `subt_ws/install` instead of `subt_ws/src` (or edit them in `src` and call a `catkin_make`).

Depending on your use-case, it might however be better to launch the robot using the official instructions at https://github.com/osrf/subt/wiki/Catkin%20System%20Setup and launch the additional CHAMP nodes calling the roslaunch commands mentioned here:

https://github.com/osrf/subt/blob/8c1e3392ef9ea88410ac201e494476e07cf9f51f/submitted_models/bosdyn_spot/launch/example.ign#L182-L192

lichenchen1205, 2021-12-23:
> Hi, if you followed the official install instructions, you should edit the files in `subt_ws/install` instead of `subt_ws/src` (or edit them in `src` and call a `catkin_make`).
>
> Depending on your [...]

thanks so much,now I can change the world,another question ，how can i get the ground truth of robots ? I edit the common.rb and set  <publish_nested_model_pose>true</publish_nested_model_pose>,but when I call roslaunch --wait subt_spot champ.launch name:=X1，the tf status was warnning in rviz.

peci1, 2021-12-26:
I don't think this kind of ground truth publishing is compatible with subt_spot. The `<publish_nested_model_pose>true</publish_nested_model_pose>` generates transform `world->X1`, and subt_spot publishes the odometry, which is `odom->X1`. Thus, X1 frame has two parents. You'll have to find a different way of processing the ground truth. We are currently using a custom Ignition-ROS program that subscribes `/world/$WORLD/pose/info` ign topic and publishes the inverse transform as `X1->X1_ground_truth`. This is not logically correct, but it works and allows having `odom` as the parent of X1. It also works around the slowness of the ground truth transform (that is fixed by #759, which has not yet been merged). Another option is to not publish a TF, but just an odom message.

peci1, 2021-12-26:
You can also try to edit champ.launch from subt_spot and disable the odometry nodes if you only need the ground truth pose.

lichenchen1205, 2021-12-27:
> You can also try to edit champ.launch from subt_spot and disable the odometry nodes if you only need the ground truth pose.

In fact, I need both the odometer and the ground truth. I think it is possible to record world->odom when the robot starts, and publish it as static tf.