osrf/subt #1046: Can not move the spot

lichenchen1205, 2021-12-01:
[image]
Thank you very much for your help. I can run this command according to your method, **but when I start it, the robot will always lie there, as shown in the picture.and I use`roslaunch champ_teleop teleop.launch`，but I can’t use the keyboard to move it**.

> You're right, that is an error in the models. I sent a fix for this to the `master` branch (#1045). With this fix, I can launch the command you mentioned correctly.

_Originally posted by @lichenchen1205 in https://github.com/osrf/subt/issues/1044#issuecomment-983376975_

lichenchen1205, 2021-12-01:
As shown in the figure, the robot has been lying there after the simulation is started, not knowing how to move it.

peci1, 2021-12-01:
Can you post the launch output? When I run the command you mentioned in #1044, the robot is standing. Are you sure you built the subt_spot package correctly?

lichenchen1205, 2021-12-01:
> 

 LC_ALL=C ign launch -v 4 ~/subt_ws/src/subt/submitted_models/explorer_spot_sensor_config_1/launch/example.ign robotName:=X1 ros:=true champ:=true
WARNING: disk usage in log directory [/home/lichenchen/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.
WARNING: disk usage in log directory [/home/lichenchen/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.
WARNING: disk usage in log directory [/home/lichenchen/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.
[Dbg] [Manager.cc:818] Forked a process for [ros] command[roslaunchsubt_roscompetition_init.launchworld_name:=examplevehicle_topics:=0enable_ground_truth:=0robot_names:=X1]
[Dbg] [Manager.cc:818] Forked a process for [topics] command[roslaunch--waitexplorer_spot_sensor_config_1vehicle_topics.launchworld_name:=examplename:=X1breadcrumbs:=0]
[Dbg] [Manager.cc:818] Forked a process for [champ] command[roslaunch--waitsubt_spotchamp.launchname:=X1]
[Dbg] [Manager.cc:1137] Loading plugin. Name[ignition::launch::GazeboServer] File[ignition-launch-gazebo]
[Msg] Logging to [/home/lichenchen/.ignition/gazebo/log/2021-12-01T20:08:49.724941525]
[Msg] Loading SDF world file[/home/lichenchen/subt_ws/install/share/bosdyn_spot/worlds/example.sdf].
[Msg] PerformerDetector publishing messages on [/subt_performer_detector]
[Dbg] [SimulationRunner.cc:891] Loaded system [ignition::gazebo::systems::PerformerDetector] for entity [4]
[Msg] Loaded level [3]
[Dbg] [Physics.cc:774] Loaded [ignition::physics::dartsim::Plugin] from library [/usr/lib/x86_64-linux-gnu/ign-physics-5/engine-plugins/libignition-physics-dartsim-plugin.so]
[Dbg] [SimulationRunner.cc:891] Loaded system [ignition::gazebo::systems::Physics] for entity [1]
[Dbg] [Sensors.cc:397] Configuring Sensors system
[Dbg] [Sensors.cc:327] SensorsPrivate::Run
[Dbg] [SimulationRunner.cc:891] Loaded system [ignition::gazebo::systems::Sensors] for entity [1]
[Dbg] [Sensors.cc:307] SensorsPrivate::RenderThread started
[Dbg] [Sensors.cc:196] Waiting for init
[Msg] Create service on [/world/example/create]
[Msg] Remove service on [/world/example/remove]
[Msg] Pose service on [/world/example/set_pose]
[Msg] Light configuration service on [/world/example/light_config]
[Msg] Physics service on [/world/example/set_physics]
[Msg] SphericalCoordinates service on [/world/example/set_spherical_coordinates]
[Msg] Enable collision service on [/world/example/enable_collision]
[Msg] Disable collision service on [/world/example/disable_collision]
[Msg] Material service on [/world/example/visual_config]
[Dbg] [SimulationRunner.cc:891] Loaded system [ignition::gazebo::systems::UserCommands] for entity [1]
[Dbg] [SimulationRunner.cc:891] Loaded system [ignition::gazebo::systems::SceneBroadcaster] for entity [1]
[Dbg] [SimulationRunner.cc:891] Loaded system [ignition::gazebo::systems::Imu] for entity [1]
[Dbg] [SimulationRunner.cc:891] Loaded system [ignmaster has started, initiating launch
... logging to /home/lichenchen/.ros/log/6ec346be-529c-11ec-9b0b-0013ef1000bd/roslaunch-lichenchen-MS-7C82-10663.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt

started roslaunch server http://lichenchen-MS-7C82:34567/

SUMMARY
========

PARAMETERS
 * /X1/base_to_footprint_ekf/base_link_frame: root
 * /X1/base_to_footprint_ekf/frequency: 50
 * /X1/base_to_footprint_ekf/odom_frame: base_footprint
 * /X1/base_to_footprint_ekf/pose0: base_to_footprint...
 * /X1/base_to_footprint_ekf/pose0_config: [True, True, True...
 * /X1/base_to_footprint_ekf/tf_prefix: X1
 * /X1/base_to_footprint_ekf/transform_time_offset: 0.045
 * /X1/base_to_footprint_ekf/transform_timeout: 0.01
 * /X1/base_to_footprint_ekf/two_d_mode: False
 * /X1/base_to_footprint_ekf/world_frame: base_footprint
 * /X1/champ_controller/gazebo: True
 * /X1/champ_controller/joint_controller_topic: joint_group_posit...
 * /X1/champ_controller/loop_rate: 250
 * /X1/champ_controller/publish_foot_contacts: False
 * /X1/champ_controller/publish_joint_control: True
 * /X1/champ_controller/publish_joint_states: False
 * /X1/footprint_to_odom_ekf/base_link_frame: base_footprint
 * /X1/footprint_to_odom_ekf/frequency: 50
 * /X1/footprint_to_odom_ekf/imu0: imu/data
 * /X1/footprint_to_odom_ekf/imu0_config: [False, False, Fa...
 * /X1/footprint_to_odom_ekf/odom0: odom/raw
 * /X1/footprint_to_odom_ekf/odom0_config: [False, False, Fa...
[... 606 lines cut ...]
[Dbg] [CameraSensor.cc:524] Camera info for [X1::camera_frontright::frontright_fisheye_image] advertised on [/world/example/model/X1/link/camera_frontright/sensor/frontright_fisheye_image/camera_info]
[Dbg] [CameraSensor.cc:271] Camera images for [X1::camera_left::left_fisheye_image] advertised on [world/example/model/X1/link/camera_left/sensor/left_fisheye_image/image]
[Dbg] [CameraSensor.cc:524] Camera info for [X1::camera_left::left_fisheye_image] advertised on [/world/example/model/X1/link/camera_left/sensor/left_fisheye_image/camera_info]
[Dbg] [CameraSensor.cc:271] Camera images for [X1::camera_right::right_fisheye_image] advertised on [world/example/model/X1/link/camera_right/sensor/right_fisheye_image/image]
[Dbg] [CameraSensor.cc:524] Camera info for [X1::camera_right::right_fisheye_image] advertised on [/world/example/model/X1/link/camera_right/sensor/right_fisheye_image/camera_info]
[... cut ...]
[Dbg] [DepthCameraSensor.cc:304] Points for [X1::camera_right::right_depth] advertised on [world/example/model/X1/link/camera_right/sensor/right_depth/depth_image/points]
[Dbg] [RgbdCameraSensor.cc:222] RGB images for [X1::body::front_camera] advertised on [world/example/model/X1/link/body/sensor/front_camera/image]
[Dbg] [RgbdCameraSensor.cc:236] Depth images for [X1::body::front_camera] advertised on [world/example/model/X1/link/body/sensor/front_camera/depth_image]
[Dbg] [RgbdCameraSensor.cc:250] Points for [X1::body::front_camera] advertised on [world/example/model/X1/link/body/sensor/front_camera/points]

here is the output,i have built the subt_spot

peci1, 2021-12-01:
I see two problems there:

```
[Err] [SystemLoader.cc:101] Failed to load system plugin [subt::GasDetector] : system not found in library [libGasEmitterDetectorPlugin.so] from path [/home/lichenchen/subt_ws/install/lib/libGasEmitterDetectorPlugin.so].
[Err] [SystemLoader.cc:101] Failed to load system plugin [subt::ConfigurableJointPositionController] : system not found in library [configurable-joint-position-controller-system] from path [/home/lichenchen/subt_ws/install/lib/libconfigurable-joint-position-controller-system.so].
```

This means that when you're launching the simulation, the libraries coming from the subt repository are not available. Did you correctly source the `install/setup.bash` of the subt simulator?

```
libEGL warning: DRI2: failed to create dri screen
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: DRI2: failed to create dri screen
```

This means you have problems with GPU. Is that something you are aware of?

lichenchen1205, 2021-12-01:
> I see two problems there:
>
> ```
> [Err] [SystemLoader.cc:101] Failed to load system plugin [subt::GasDetector] : system not found in library [libGasEmitterDetectorPlugin.so] from path [...]

Thank you for your patient guidance and I have already recompile the wokspace and source the install/setup.bash  But the problem still arises, even if it is implemented ign launch -v 4 competition.ign worldName:=tunnel_circuit_practice_01 circuit:=tunnel robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1
with output `[Err] [SystemLoader.cc:101] Failed to load system plugin [subt::GameLogicPlugin] : system not found in library  [libGameLogicPlugin.so] from path [/home/lichenchen/subt_ws/install/lib/libGameLogicPlugin.so]` and i can found the libGameLogicPlugin.so in the path

peci1, 2021-12-01:
Please post the output of `echo $LD_LIBRARY_PATH` and `echo $IGN_GAZEBO_SYSTEM_PLUGIN_PATH`.

lichenchen1205, 2021-12-01:
> Please post the output of `echo $LD_LIBRARY_PATH` and `echo $IGN_GAZEBO_SYSTEM_PLUGIN_PATH`.

[image]
The result is shown in the figure

peci1, 2021-12-01:
That seems correct. Could you go once more through the setup guide and check that you did everything correctly? Did catkin_make report any errors?

Also, please, post the output of `ldd ~/subt_ws/install/lib/libGameLogicPlugin.so`.

lichenchen1205, 2021-12-03:
> That seems correct. Could you go once more through the setup guide and check that you did everything correctly? Did catkin_make report any errors?
> 
> Also, please, post the output of `ldd ~/subt_ws/install/lib/libGameLogicPlugin.so`.

Thank you very much for your help. I uninstalled gazebo and recompiled the program. Now I can move the robot, but it seems that the robot is easy to fall. Is this normal?

peci1, 2021-12-03:
Yes, it is. The motion model is pretty simplistic (although CHAMP - the used model - has put a lot of effort into generating good motion patterns, it is nowhere near the motion controller Boston Dynamics provides for the real Spot).

One potentially easy way to improve the motion on inclined surfaces could be to add a whole-body motion controller that would try to level the body no matter what the inclination of the terrain is.

Share your improvements if you have some!

lichenchen1205, 2021-12-03:
Okay, there is one more question, what level of accuracy can the robot's odometer reach?

peci1, 2021-12-03:
On flat ground, the kinematic model is used to estimate the odometry, which could be precise enough until the robot starts slipping. On uneven ground or when slipping, I wouldn't trust it too much. It is always a good idea to combine the legged odometry with IMU.

lichenchen1205, 2021-12-03:
okay，thank you for your help ，I have no questions

closed 2021-12-03 by lichenchen1205