osrf/subt #1049: Unable to create the rendering window

rm-azou, 2022-02-16:
When I run the commond in the Hello World([[Tutorial 1](https://github.com/osrf/subt_hello_world/blob/master/posts/01_hello_world.md)]) on a remote server to tun the subt docker container
`sudo ./subt/docker/run.bash osrf/subt-virtual-testbed:latest cave_circuit.ign worldName:=simple_cave_01 robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1 circuit:=cave headless:=true`,
there is an error: 
`process[X1/ros_ign_bridge_gpu_lidar-11]: started with pid [760]
[Err] [Ogre2RenderEngine.cc:744]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXContext.cpp (line 61)
[Err] [Ogre2RenderEngine.cc:744]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXContext.cpp (line 61)
process[X1/ros_ign_bridge_front_laser_set_rate-12]: started with pid [858]
[Err] [Ogre2RenderEngine.cc:744]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXContext.cpp (line 61)
[Err] [Ogre2RenderEngine.cc:744]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXContext.cpp (line 61)
process[X1/ros_ign_bridge_odom-13]: started with pid [890]
[Err] [Ogre2RenderEngine.cc:744]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXContext.cpp (line 61)
[Err] [Ogre2RenderEngine.cc:744]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXContext.cpp (line 61)
process[X1/ros_ign_bridge_battery_state-14]: started with pid [920]
[Err] [Ogre2RenderEngine.cc:744]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXContext.cpp (line 61)
[Err] [Ogre2RenderEngine.cc:744]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXContext.cpp (line 61)
process[X1/pose_tf_broadcaster-15]: started with pid [951]
[Err] [Ogre2RenderEngine.cc:744]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXContext.cpp (line 61)
[Err] [Ogre2RenderEngine.cc:744]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXContext.cpp (line 61)
[Err] [Ogre2RenderEngine.cc:752] Unable to create the rendering window after [11] attempts.
[Err] [Ogre2RenderEngine.cc:681] Failed to create dummy render window.`

[image]
[image]

peci1, 2022-02-16:
That's expected. The SubT simulator needs a running X server. Your machine shows none. You can try running the simulator inside `xvfb` to at least make it running with software rendering.

rm-azou, 2022-02-17:
> That's expected. The SubT simulator needs a running X server. Your machine shows none. You can try running the simulator inside `xvfb` to at least make it running with software rendering.

Thank you! I have solved this problem using your sugestion. It really helps me a lot, I can finally use simulator on my GPU server.