osrf/subt #1057: How to Create a Controllable LED on a Robot in SubT

Space-Swarm, 2024-02-15:
Hi everyone, 

I'm looking to create a controllable LED on a robot in the SubT simulator for navigational purposes with **Ignition Dome**.

I am looking for the LED to have the **following functionality**:
1. Controllable on/off for the LED via ros/cpp
2. A way to set the colour of the LED via ros/cpp

Does anyone have any advice on how to go about doing this? If any of the teams have done this in the past, please do let me know how you went about it.

**Further Information**
I am currently looking at editing the **Marble Husky model** (sensor config 1) to include this, potentially by editing the SDF file & the vehicle topics launch file: 
SDF FIle - https://github.com/osrf/subt/blob/master/submitted_models/marble_husky_sensor_config_1/model.sdf
Vehicle Topics Launch File - https://github.com/osrf/subt/blob/master/submitted_models/marble_husky_sensor_config_1/launch/vehicle_topics.launch

I was thinking if there was a way to overcome the cross compatibility issue, I could integrate this **gazebo classic plugin**, but that looks tricky: https://classic.gazebosim.org/tutorials?tut=led_plugin&cat=plugins 

**Things I have already tried:**
I tried editing the **DAE file** for the spotlight that the model already has, but it didn't change the colour and even if it did, it wouldn't be controllable like I need it - https://github.com/osrf/subt/blob/master/submitted_models/marble_husky_sensor_config_1/meshes/spotlight.dae

I have found a number of **existing models** use different lights as part of the model, so perhaps one of these might help too:
Two LEDS (multiple models use this) - https://github.com/osrf/subt/blob/master/submitted_models/marble_spot_sensor_config_1/urdf/lights.xacro
Hexapod LED stripe - https://github.com/osrf/subt/blob/master/submitted_models/ctu_cras_norlab_lily_sensor_config_1/urdf/lights.xacro
Absolem light (only 1 I think): https://github.com/osrf/subt/blob/master/submitted_models/marble_spot_sensor_config_1/urdf/lights.xacro
Different light on spot: https://github.com/osrf/subt/blob/master/submitted_models/csiro_data61_spot_sensor_config_1/urdf/nav_backpack/sensors/light.xacro

peci1, 2024-02-15:
It might work to add the permanent lights and then a "fake link" with a controllable joint that would be able to rotate the link in a caleidoscope-like fashion. On part of the fake link can be solid black, blocking all light from the LED. Another part could be half-transparent with a color tint to change the light color. And last, there could be a part with nothing that would just let the light out unchanged.

The fake link should be an object with just a `<visual>` and no `<collision>` element. Or, if Gazebo doesn't like it, add a centimeter-sized collision and a 0.01 gram inertial element to the fake link.

Space-Swarm, 2024-02-21:
Thank you so much, that's a great suggestion. I'm looking at that now, and it looks a bit complex for my current skillset. Is there a simpler way of doing it? 

Another option I'm contemplating (however it is also complex): writing my own plugin with Ignition using code from https://gazebosim.org/api/gazebo/6/light_config.html and wrapping a ros topic into it.

I was thinking maybe Ignition Fortress has a ready made plugin for lights? It uses Gazebo 9.

xingge520, 2024-05-14:
Hello, may I ask if your [CloudSim Docker Compsose ](https://github.com/osrf/subt/wiki/Cloudsim%20Docker%20Compose) has been successfully built,Especially how to build subt_solution:latest images