osrf/subt #1062: Installation issues in May 2024

nschurr, 2024-05-10:
I am trying to install and run the SubT simulator in May 2024 following the Get Started page https://github.com/osrf/subt/wiki/Get%20Started.  Attempts with the three described methods (docker, Ignition Fortress+catkin, Ignition Dome + catkin) are failing in different ways.

Question 1: What installation procedure (if any) have people had success with recently?

Question 2: Are any of our specs below known to be incompatible?

 * Ubuntu 18.04.6
 * CUDA 12.3
 * NVIDIA Driver 545.29.06
 * Processor: 13th Gen Intel Core i7-12700F x 24
 * Graphics: NVIDIA GeForce RTX 4060/PCIe/SSE2
 * Memory: 31.2 GiB

Question 3:  In the catkin workspace method, is there a way to link the models after manually downloading them?  If so, where, and in what format, should the models be placed?  Alternatively, is there a way to change the automatic download paths to match the new location?

Brief descriptions of our steps are described below.  Guidance on getting a working installation with any of these methods would be appreciated.  Additional information or logs can be posted upon request.  Thanks!

--------------------------------------------
**Attempt procedures and their associated failure modes.**

**Method 1: Docker Instructions**

Summary: A blank black Gazebo window opens for a few seconds before closing when running the command to run the virtual testbed image.  Possibly related to OpenGL/libGL.

Steps to Reproduce:
* Follow the Docker Install instructions (https://github.com/osrf/subt/wiki/Docker%20Install).
	* Both the `sudo docker run hello-world` and `docker run --runtime=nvidia --rm nvidia/cuda:12.0.0-base-ubuntu18.04 nvidia-smi` checks appear to run as expected.
* Follow the Docker System Setup instructions (https://github.com/osrf/subt/wiki/Docker%20System%20Setup)
	* Fails to run `./run.bash osrf/subt-virtual-testbed competition.ign worldName:=tunnel_circuit_practice_01 circuit:=tunnel robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1`

Observed errors:
* Red error message: `[GUI] [Err] [Application.cc:673] [QT] Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior QSurfaceFormat::SwapBehavior(DoubleBuffer), swapInterval 1, profile  QSurfaceFormat::OpenGLContextProfile(NoProfile))`
* Also white error messages `libGL error: No matching fbConfigs or visuals found` and `libGL error: failed to load driver: swrast`.

**Method 2: Ignition Fortress**

Summary:  Cannot apt-get install ignition fortress - broken packages

Steps to Reproduce:  
* Follow steps on https://github.com/osrf/subt/wiki/Fortress
* Fails to run `sudo apt-get -y install ignition-fortress`

Observed errors:

```msi-ub18-subt@msi-ub18-subt:~/subt_testbed$ sudo apt-get -y install ignition-fortress
[sudo] password for msi-ub18-subt: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:
The following packages have unmet dependencies:
 ignition-fortress : Depends: python3-ignition-gazebo6 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Additional information:
* Following the Gazebo instructions on https://gazebosim.org/docs/fortress/install_ubuntu results in the same error.

**Method 3:  Ignition Dome + Catkin**

Summary: SubT models appear to have moved and are not retrieved automatically

Steps to reproduce:
* Follow steps at https://github.com/osrf/subt/wiki/Catkin%20System%20Setup
* Step 2 note:  used `--include-eol-distros` with `rosdep update`
* Step 4 note: several messages of `Cannot locate rosdep definition for [...]` ignition packages, but confirmed their presence with `dpkg -l` as in System Requirements page (https://github.com/osrf/subt/wiki/System%20Requirements)
* Step 5 note:  Needed to install PCL and associated packages separately.
* Step 5 failure:
```
  [Err] [FuelClient.cc:669] Failed to download model.
  Server: https://fuel.ignitionrobotics.org
  Route: openrobotics/models/subt_tunnel_staging_area/tip/subt_tunnel_staging_area.zip
  REST response code: 301
```

When allowed to run longer, the message as in https://github.com/osrf/subt/issues/1033 repeats.

The models do not appear to be located at https://fuel.ignitionrobotics.org/openrobotics/models/subt_tunnel_staging_area anymore.  They appear to be at https://app.gazebosim.org/OpenRobotics/fuel/collections/SubT%20Tech%20Repo (e.g., https://app.gazebosim.org/OpenRobotics/fuel/models/subt_tunnel_staging_area)

nschurr, 2024-06-11:
Update:
Trying Method 3 (Dome + Catkin) again about a month later, with no changes, the automatic download connects and retrieves models.

peci1, 2024-06-11:
Thanks for reporting the partial success. We're now working on Fortress install instructions, which should include reworking the Fuel models. Should I ping you when the installation guide is ready?

Ketan13294, 2024-12-12:
Has there been any update on this?

peci1, 2024-12-12:
Yes, you can try the new experimental installation guide: https://github.com/osrf/subt/wiki/Installation-for-ROS-Noetic .

closed 2025-01-31 by nschurr