osrf/subt #1063: Aborted (core dumped) when test installation

jiaminguo1995, 2024-12-04:
I try this project recently, but when I run the code " ign launch -v 4 competition.ign worldName:=tunnel_circuit_practice_01 circuit:=tunnel robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1 levels:=false" to test installation, the terminal will shut down.
here are the error

terminate called after throwing an instance of 'Ogre::InvalidParametersException'
  what():  OGRE EXCEPTION(2:InvalidParametersException): All framebuffer formats with this texture internal format unsupported in GL3PlusFrameBufferObject::initialise at /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusFrameBufferObject.cpp (line 229)
Stack trace (most recent call last) in thread 4326:
#29   Object "", at 0xffffffffffffffff, in 
[GUI] [Wrn] [Component.hh:189] Trying to deserialize component with data type [St6vectorIdSaIdEE], which doesn't have `operator>>`. Component will not be deserialized.
[GUI] [Dbg] [EntityComponentManager.cc:767] Using components of type [5681045647686850796] / [ign_gazebo_components.SlipComplianceCmd ].
#28   Source "/build/glibc-CVJwZb/glibc-2.27/misc/../sysdeps/unix/sysv/linux/x86_64/clone.S", line 95, in  [0x7f38956dd61e]
#27   Source "/build/glibc-CVJwZb/glibc-2.27/nptl/pthread_create.c", line 463, in start_thread [0x7f38953a46da]
#26   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f3892f866de, in std::error_code::default_error_condition() const
#25   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-4/plugins/libignition-gazebo-sensors-system.so", at 0x7f385642ec6f, in ignition::gazebo::v4::systems::SensorsPrivate::RenderThread()
#24   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-4/plugins/libignition-gazebo-sensors-system.so", at 0x7f385642e737, in ignition::gazebo::v4::systems::SensorsPrivate::RunOnce()
#23   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-4/engine-plugins/libignition-rendering-ogre2.so", at 0x7f37f452bf6f, in ignition::rendering::v4::Ogre2Scene::PreRender()
#22   Object "/usr/lib/x86_64-linux-gnu/libignition-rendering4.so.4", at 0x7f385581ad7b, in ignition::rendering::v4::BaseScene::PreRender()
#21   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-4/engine-plugins/libignition-rendering-ogre2.so", at 0x7f37f44ac5a5, in ignition::rendering::v4::BaseVisual<ignition::rendering::v4::Ogre2Node>::PreRender()
#20   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-4/engine-plugins/libignition-rendering-ogre2.so", at 0x7f37f44b1561, in ignition::rendering::v4::BaseVisual<ignition::rendering::v4::Ogre2Node>::PreRenderChildren()
#19   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-4/engine-plugins/libignition-rendering-ogre2.so", at 0x7f37f44ac5a5, in ignition::rendering::v4::BaseVisual<ignition::rendering::v4::Ogre2Node>::PreRender()
#18   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-4/engine-plugins/libignition-rendering-ogre2.so", at 0x7f37f44b1561, in ignition::rendering::v4::BaseVisual<ignition::rendering::v4::Ogre2Node>::PreRenderChildren()
#17   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-4/engine-plugins/libignition-rendering-ogre2.so", at 0x7f37f44ac5a5, in ignition::rendering::v4::BaseVisual<ignition::rendering::v4::Ogre2Node>::PreRender()
#16   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-4/engine-plugins/libignition-rendering-ogre2.so", at 0x7f37f44b1561, in ignition::rendering::v4::BaseVisual<ignition::rendering::v4::Ogre2Node>::PreRenderChildren()
#15   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-4/engine-plugins/libignition-rendering-ogre2.so", at 0x7f37f44e7278, in ignition::rendering::v4::Ogre2GpuRays::CreateGpuRaysTextures()
#14   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-4/engine-plugins/libignition-rendering-ogre2.so", at 0x7f37f44e2cab, in ignition::rendering::v4::Ogre2GpuRays::Setup1stPass()
#13   Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f37d840da64, in Ogre::TextureManager::createManual(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Ogre::TextureType, unsigned int, unsigned int, unsigned int, int, Ogre::PixelFormat, int, Ogre::ManualResourceLoader*, bool, unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool)
#12   Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f37d840954c, in Ogre::Texture::createInternalResources()
#11   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7f36941a74b7, in Ogre::GL3PlusTexture::createInternalResourcesImpl()
#10   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7f36941a69ef, in Ogre::GL3PlusTexture::_createSurfaceList()
#9    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7f36941aaedd, in Ogre::v1::GL3PlusTextureBuffer::GL3PlusTextureBuffer(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, int, int, Ogre::v1::HardwareBuffer::Usage, bool, unsigned int)
#8    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7f3694180eb1, in Ogre::GL3PlusFBOManager::createRenderTexture(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Ogre::GL3PlusSurfaceDesc const&, bool, unsigned int)
#7    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7f3694180e25, in Ogre::GL3PlusFBORenderTexture::GL3PlusFBORenderTexture(Ogre::GL3PlusFBOManager*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Ogre::GL3PlusSurfaceDesc const&, bool, unsigned int)
#6    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7f369418408e, in Ogre::GL3PlusFrameBufferObject::initialise()
#5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f3892f5bd53, in __cxa_throw
#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f3892f5bb20, in std::terminate()
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f3892f5bae5, in std::rethrow_exception(std::__exception_ptr::exception_ptr)
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f3892f55956, in 
#1    Source "/build/glibc-CVJwZb/glibc-2.27/stdlib/abort.c", line 79, in __GI_abort [0x7f38955fc7f0]
#0    Source "/build/glibc-CVJwZb/glibc-2.27/signal/../sysdeps/unix/sysv/linux/raise.c", line 51, in __GI_raise [0x7f38955fae87]
Aborted (Signal sent by tkill() 3408 1000)
Aborted (core dumped)

Thanks for everyone who can give me some advice for this problem.

peci1, 2024-12-04:
That looks like a problem with your GPU. What GPU do you have? Do you run in any kind of virtualization?

jiaminguo1995, 2024-12-04:
> That looks like a problem with your GPU. What GPU do you have? Do you run in any kind of virtualization?

My GPU id geforce rtx 2060 and I do not run in virtualization.

jiaminguo1995, 2024-12-04:
> That looks like a problem with your GPU. What GPU do you have? Do you run in any kind of virtualization?
I try the robot X3, it can work. But the RTF is only 20%

peci1, 2024-12-04:
> All framebuffer formats with this texture internal format unsupported

This error indicates that your GPU doesn't support some requested texture format. With 2060, that's weird. One cause could be drivers. Do you use the latest Nvidia binary driver? Please, post the terminal output of `glxgears -info`. If you don't have glxgears, install package mesa-utils. Also, please attach the output of nvidia-smi . And what is your operating system and method of installation?

---

So you got the simulation running with X3 robot, but not X1? Anyways, SubT simulator is not a 3D shooter game. Do not expect real-time performance. Not even on a high-end GPU. The simulator is not optimized for that. It is constructed for simulation of autonomous behaviors and you have to be prepared that the simulations will take their time. Just to give you a glimpse, in contrary to traditional 3D shooters which render 1 FullHd or 4K camera, each robot in SubT simulator needs about 4-10 cameras rendered. That's just one of the many reasons why SubT simulator will hardly ever run real-time. The RTF also depends on the loaded world and robot (its sensors and number of terrain contact points).

jiaminguo1995, 2024-12-04:
Thank you so much. Here are the output of glxgears -info and nvidia-smi.

jiaminguo1995, 2024-12-04:
> This error indicates that your GPU doesn't support some requested texture format. With 2060, that's weird. One cause could be drivers. Do you use the latest Nvidia binary driver? Please, post the [...]

Thank you so much.  Here are the output of glxgears -info and nvidia-smi. 
[image]
[image]
My operating system is 18.04.  I installed this system followed by this link: https://jingyan.baidu.com/article/154b46312f033e28ca8f41c2.html.

It is strange that X3 and X4 can work, but other robots do not work. It seems that only the drone is available on my computer, but my research needs to worked on wheeled robots or legged robots.

peci1, 2024-12-04:
The interesting info is actually at the beginning of glxgears -info, not at its end. Please scroll up and send it again. Even better, send the whole output as text.

I guess the reason why some robots work and some don't would be the sensors. Depth cameras and lidars require some less traditional texture types. You can experiment with some other sensor configs for the ground robots.

peci1, 2024-12-04:
Also there should be file ~/.ignition/rendering/ogre.log . It can also contain important information. Please, copy paste it here.

jiaminguo1995, 2024-12-05:
> The interesting info is actually at the beginning of glxgears -info, not at its end. Please scroll up and send it again. Even better, send the whole output as text.
>
> I guess the reason why some [...]
Thank you for your answer. Here is the output of glxgears -info:
jamie@jamie:~$ glxgears -info
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
GL_RENDERER   = Mesa DRI Intel(R) UHD Graphics (CML GT2)
GL_VERSION    = 3.0 Mesa 20.0.8
GL_VENDOR     = Intel Open Source Technology Center
GL_EXTENSIONS = GL_ARB_multisample GL_EXT_abgr GL_EXT_bgra GL_EXT_blend_color GL_EXT_blend_minmax GL_EXT_blend_subtract GL_EXT_copy_texture GL_EXT_subtexture GL_EXT_texture_object GL_EXT_vertex_array GL_EXT_compiled_vertex_array GL_EXT_texture GL_EXT_texture3D GL_IBM_rasterpos_clip GL_ARB_point_parameters GL_EXT_draw_range_elements GL_EXT_packed_pixels GL_EXT_point_parameters GL_EXT_rescale_normal GL_EXT_separate_specular_color GL_EXT_texture_edge_clamp GL_SGIS_generate_mipmap GL_SGIS_texture_border_clamp GL_SGIS_texture_edge_clamp GL_SGIS_texture_lod GL_ARB_framebuffer_sRGB GL_ARB_multitexture GL_EXT_framebuffer_sRGB GL_IBM_multimode_draw_arrays GL_IBM_texture_mirrored_repeat GL_3DFX_texture_compression_FXT1 GL_ARB_texture_cube_map GL_ARB_texture_env_add GL_ARB_transpose_matrix GL_EXT_blend_func_separate GL_EXT_fog_coord GL_EXT_multi_draw_arrays GL_EXT_secondary_color GL_EXT_texture_env_add GL_EXT_texture_filter_anisotropic GL_EXT_texture_lod_bias GL_INGR_blend_func_separate GL_NV_blend_square GL_NV_light_max_exponent GL_NV_texgen_reflection GL_NV_texture_env_combine4 GL_S3_s3tc GL_SUN_multi_draw_arrays GL_ARB_texture_border_clamp GL_ARB_texture_compression GL_EXT_framebuffer_object GL_EXT_texture_compression_s3tc GL_EXT_texture_env_combine GL_EXT_texture_env_dot3 GL_MESA_window_pos GL_NV_packed_depth_stencil GL_NV_texture_rectangle GL_ARB_depth_texture GL_ARB_occlusion_query GL_ARB_shadow GL_ARB_texture_env_combine GL_ARB_texture_env_crossbar GL_ARB_texture_env_dot3 GL_ARB_texture_mirrored_repeat GL_ARB_window_pos GL_EXT_stencil_two_side GL_EXT_texture_cube_map GL_NV_depth_clamp GL_NV_fog_distance GL_APPLE_packed_pixels GL_ARB_draw_buffers GL_ARB_fragment_program GL_ARB_fragment_shader GL_ARB_shader_objects GL_ARB_vertex_program GL_ARB_vertex_shader GL_ATI_draw_buffers GL_ATI_texture_env_combine3 GL_ATI_texture_float GL_EXT_shadow_funcs GL_EXT_stencil_wrap GL_MESA_pack_invert GL_NV_primitive_restart GL_ARB_depth_clamp GL_ARB_fragment_program_shadow GL_ARB_half_float_pixel GL_ARB_occlusion_query2 GL_ARB_point_sprite GL_ARB_shading_language_100 GL_ARB_sync GL_ARB_texture_non_power_of_two GL_ARB_vertex_buffer_object GL_ATI_blend_equation_separate GL_EXT_blend_equation_separate GL_OES_read_format GL_ARB_color_buffer_float GL_ARB_pixel_buffer_object GL_ARB_texture_compression_rgtc GL_ARB_texture_float GL_ARB_texture_rectangle GL_EXT_packed_float GL_EXT_pixel_buffer_object GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_rectangle GL_EXT_texture_sRGB GL_EXT_texture_shared_exponent GL_ARB_framebuffer_object GL_EXT_framebuffer_blit GL_EXT_framebuffer_multisample GL_EXT_packed_depth_stencil GL_APPLE_object_purgeable GL_ARB_vertex_array_object GL_ATI_separate_stencil GL_EXT_draw_buffers2 GL_EXT_draw_instanced GL_EXT_gpu_program_parameters GL_EXT_gpu_shader4 GL_EXT_texture_array GL_EXT_texture_integer GL_EXT_texture_sRGB_decode GL_EXT_timer_query GL_OES_EGL_image GL_AMD_texture_texture4 GL_ARB_copy_buffer GL_ARB_depth_buffer_float GL_ARB_draw_instanced GL_ARB_half_float_vertex GL_ARB_instanced_arrays GL_ARB_map_buffer_range GL_ARB_texture_rg GL_ARB_texture_swizzle GL_ARB_vertex_array_bgra GL_EXT_texture_swizzle GL_EXT_vertex_array_bgra GL_NV_conditional_render GL_AMD_conservative_depth GL_AMD_depth_clamp_separate GL_AMD_draw_buffers_blend GL_AMD_seamless_cubemap_per_texture GL_AMD_shader_stencil_export GL_ARB_ES2_compatibility GL_ARB_blend_func_extended GL_ARB_debug_output GL_ARB_draw_buffers_blend GL_ARB_draw_elements_base_vertex GL_ARB_explicit_attrib_location GL_ARB_fragment_coord_conventions GL_ARB_provoking_vertex GL_ARB_sample_shading GL_ARB_sampler_objects GL_ARB_seamless_cube_map GL_ARB_shader_stencil_export GL_ARB_shader_texture_lod GL_ARB_texture_cube_map_array GL_ARB_texture_gather GL_ARB_texture_multisample GL_ARB_texture_query_lod GL_ARB_texture_rgb10_a2ui GL_ARB_uniform_buffer_object GL_ARB_vertex_type_2_10_10_10_rev GL_EXT_provoking_vertex GL_EXT_texture_snorm GL_MESA_texture_signed_rgba GL_NV_texture_barrier GL_ARB_draw_indirect GL_ARB_get_program_binary GL_ARB_robustness GL_ARB_separate_shader_objects GL_ARB_shader_bit_encoding GL_ARB_shader_precision GL_ARB_texture_compression_bptc GL_ARB_timer_query GL_ARB_transform_feedback2 GL_ARB_transform_feedback3 GL_EXT_direct_state_access GL_AMD_multi_draw_indirect GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_ARB_compressed_texture_pixel_storage GL_ARB_conservative_depth GL_ARB_internalformat_query GL_ARB_map_buffer_alignment GL_ARB_shader_atomic_counters GL_ARB_shader_image_load_store GL_ARB_shading_language_420pack GL_ARB_shading_language_packing GL_ARB_texture_storage GL_ARB_transform_feedback_instanced GL_EXT_framebuffer_multisample_blit_scaled GL_EXT_transform_feedback GL_AMD_query_buffer_object GL_AMD_shader_trinary_minmax GL_ARB_ES3_compatibility GL_ARB_arrays_of_arrays GL_ARB_clear_buffer_object GL_ARB_compute_shader GL_ARB_copy_im [... cut ...] polygon_offset_clamp GL_ARB_texture_filter_anisotropic GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_EXT_shader_framebuffer_fetch_non_coherent GL_EXT_texture_sRGB_R8 GL_EXT_texture_shadow_lod GL_INTEL_shader_atomic_float_minmax GL_INTEL_shader_integer_functions2 GL_NV_compute_shader_derivatives GL_EXT_EGL_sync GL_EXT_demote_to_helper_invocation 
VisualID 807, 0x327
741 frames in 5.0 seconds = 147.784 FPS
721 frames in 5.0 seconds = 144.200 FPS
721 frames in 5.0 seconds = 144.071 FPS

jiaminguo1995, 2024-12-05:
> Also there should be file ~/.ignition/rendering/ogre.log . It can also contain important information. Please, copy paste it here.
In the path ~/.ignition/rendering/, I only found a file named ogre2.log. The information is as follows:

[ogre2.log](https://github.com/user-attachments/files/18018160/ogre2.log)

peci1, 2024-12-05:
Ok, now it starts to be clearer.

Gazebo does not use the NVidia card, but the integrated GPU. But your processor is too new for this version of Ubuntu (released 2021, vs. Ubuntu released 2018). So most probably, the version of Ubuntu cannot utilize all features of the CPU and its integrated GPU.

Here are your options:

1. Try `export DRI_PRIME=1` before launching Gazebo (in the same console where you launch it). I'm not sure what are the other requirements, but on some systems, it switches to the discrete GPU.
2. Install [VirtualGL](https://github.com/VirtualGL/virtualgl) and prefix the `ign launch` command with `vgl-run +v -d /dev/dri/card1 ign launch...` . If it behaves the same, try `card0` instead of `card1`. One of them should be the NVidia GPU.
3. Use [kumk](https://github.com/peci1/kumk) or any other tool to install mainline kernel, and [use a newer Mesa (GPU subsystem)](https://launchpad.net/~kisak/+archive/ubuntu/turtle), e.g. by calling `sudo add-apt-repository ppa:kisak/turtle && sudo apt update && sudo apt dist-upgrade` and rebooting your computer. Please note that this approach is the most intrusive to your system and it can break other things.

jiaminguo1995, 2024-12-05:
> Ok, now it starts to be clearer.
>
> Gazebo does not use the NVidia card, but the integrated GPU. But your processor is too new for this version of Ubuntu (released 2021, vs. Ubuntu released 2018). So [...]

Thank you so much! The problem has been solved by method 3. Method 1 does not work.  When trying method 2, I installed VirtualGL using the source code installation method. However, the installation was not successful, because the cmake .. failed.  Then I tried the method 3, it worked.  Could you please  help me for another question？ Have you tried deploying Subt on Ubuntu 20.04？ As you say, my GPU is too new for ubuntu 18, I think Ubuntu 18 is a bit old for this era.  Actually, this computer was bought 5 days ago in the second market, just for this project.  Afterwards, I started installing from the system according to Subt's requirements.  If Subt can be deployed on 20.04, it will bring great convenience to my subsequent development.  If it can be deployed, I will try to test it on Ubuntu 20.04. If not, I will keep this system unchanged. Your answer will effectively help me spend less time on testing the system.

peci1, 2024-12-05:
> I installed VirtualGL using the source code installation method. However, the installation was not successful, because the cmake .. failed.

VirtualGL is easy to install via the provided DEB archives, e.g. https://github.com/VirtualGL/virtualgl/releases/download/3.1.1/virtualgl_3.1.1_amd64.deb (after you download, just `sudo dpkg -i virtualgl_3.1.1_amd64.deb`).

> If Subt can be deployed on 20.04,

There is an experimental guide that can help you set up SubT on Ubuntu 20.04: https://github.com/osrf/subt/wiki/Installation-for-ROS-Noetic . Please, if you try it, post feedback (ideally into a new issue).

jiaminguo1995, 2024-12-06:
> VirtualGL is easy to install via the provided DEB archives, e.g. https://github.com/VirtualGL/virtualgl/releases/download/3.1.1/virtualgl_3.1.1_amd64.deb (after you download, just `sudo dpkg -i [...]

Thank you for your reply! The link helps me a lot. I'll be sure to get feedback to you when I test it on ubuntu 20. (I'll  test it on Ubuntu 20 after buying a solid-state drive during the Double 12 sale, the current system is still kept to ensure that the research can proceed smoothly.)

kudouxiao, 2024-12-07:
@peci1 

> Ok, now it starts to be clearer.
>
> Gazebo does not use the NVidia card, but the integrated GPU. But your processor is too new for this version of Ubuntu (released 2021, vs. Ubuntu released 2018). So [...]

Hello, I am using a Docker image to start up, and when running ./run.bash osrf/subt-virtual-testbed competition.ign worldName:=tunnel_circuit_practice_01 circuit:=tunnel robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1, I encounter the same issue. I have tried the first and third methods without resolution, and I do not know how to run the second method when using Docker. My system is Ubuntu 24.04, and the graphics card is similar to the previous issue, which is an RTX 2060 Max-Q.
Please help me. Thank you very much.

peci1, 2024-12-07:
@kudouxiao Please, provide the same debugging info I requested the the original issue author and specify your CPU. And please, provide the output of the debug commands both outside docker and inside.

kudouxiao, 2024-12-08:
> @kudouxiao Please, provide the same debugging info I requested the the original issue author and specify your CPU. And please, provide the output of the debug commands both outside docker and inside.请提供我向原始问题作者请求的相同调试信息并指定您的 CPU。请提供 docker 外部和内部的调试命令的输出。

**My CPU is i7-10750H.**
### 
glxgears -info:
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
GL_RENDERER   = Mesa Intel(R) UHD Graphics (CML GT2)
GL_VERSION    = 4.6 (Compatibility Profile) Mesa 24.2.8 - kisak-mesa PPA
GL_VENDOR     = Intel
GL_EXTENSIONS = GL_ARB_multisample GL_EXT_abgr GL_EXT_bgra GL_EXT_blend_color GL_EXT_blend_minmax GL_EXT_blend_subtract GL_EXT_copy_texture GL_EXT_subtexture GL_EXT_texture_object GL_EXT_vertex_array GL_EXT_compiled_vertex_array GL_EXT_texture GL_EXT_texture3D GL_IBM_rasterpos_clip GL_ARB_point_parameters GL_EXT_draw_range_elements GL_EXT_packed_pixels GL_EXT_point_parameters GL_EXT_rescale_normal GL_EXT_separate_specular_color GL_EXT_texture_edge_clamp GL_SGIS_generate_mipmap GL_SGIS_texture_border_clamp GL_SGIS_texture_edge_clamp GL_SGIS_texture_lod GL_ARB_framebuffer_sRGB GL_ARB_multitexture GL_EXT_framebuffer_sRGB GL_IBM_multimode_draw_arrays GL_IBM_texture_mirrored_repeat GL_3DFX_texture_compression_FXT1 GL_ARB_texture_cube_map GL_ARB_texture_env_add GL_ARB_transpose_matrix GL_EXT_blend_func_separate GL_EXT_fog_coord GL_EXT_multi_draw_arrays GL_EXT_secondary_color GL_EXT_texture_env_add GL_EXT_texture_filter_anisotropic GL_EXT_texture_lod_bias GL_INGR_blend_func_separate GL_NV_blend_square GL_NV_light_max_exponent GL_NV_texgen_reflection GL_NV_texture_env_combine4 GL_S3_s3tc GL_SUN_multi_draw_arrays GL_ARB_texture_border_clamp GL_ARB_texture_compression GL_EXT_framebuffer_object GL_EXT_texture_compression_s3tc GL_EXT_texture_env_combine GL_EXT_texture_env_dot3 GL_MESA_window_pos GL_NV_packed_depth_stencil GL_NV_texture_rectangle GL_ARB_depth_texture GL_ARB_occlusion_query GL_ARB_shadow GL_ARB_texture_env_combine GL_ARB_texture_env_crossbar GL_ARB_texture_env_dot3 GL_ARB_texture_mirrored_repeat GL_ARB_window_pos GL_ATI_fragment_shader GL_EXT_stencil_two_side GL_EXT_texture_cube_map GL_NV_copy_depth_to_color GL_NV_depth_clamp GL_NV_fog_distance GL_NV_half_float GL_APPLE_packed_pixels GL_ARB_draw_buffers GL_ARB_fragment_program GL_ARB_fragment_shader GL_ARB_shader_objects GL_ARB_vertex_program GL_ARB_vertex_shader GL_ATI_draw_buffers GL_ATI_texture_env_combine3 GL_ATI_texture_float GL_EXT_shadow_funcs GL_EXT_stencil_wrap GL_MESA_pack_invert GL_NV_primitive_restart GL_ARB_depth_clamp GL_ARB_fragment_program_shadow GL_ARB_half_float_pixel GL_ARB_occlusion_query2 GL_ARB_point_sprite GL_ARB_shading_language_100 GL_ARB_sync GL_ARB_texture_non_power_of_two GL_ARB_vertex_buffer_object GL_ATI_blend_equation_separate GL_EXT_blend_equation_separate GL_OES_read_format GL_ARB_color_buffer_float GL_ARB_pixel_buffer_object GL_ARB_texture_compression_rgtc GL_ARB_texture_float GL_ARB_texture_rectangle GL_ATI_texture_compression_3dc GL_EXT_packed_float GL_EXT_pixel_buffer_object GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_rectangle GL_EXT_texture_sRGB GL_EXT_texture_shared_exponent GL_ARB_framebuffer_object GL_EXT_framebuffer_blit GL_EXT_framebuffer_multisample GL_EXT_packed_depth_stencil GL_ARB_vertex_array_object GL_ATI_separate_stencil GL_EXT_draw_buffers2 GL_EXT_draw_instanced GL_EXT_gpu_program_parameters GL_EXT_gpu_shader4 GL_EXT_texture_array GL_EXT_texture_compression_latc GL_EXT_texture_integer GL_EXT_texture_sRGB_decode GL_EXT_timer_query GL_OES_EGL_image GL_AMD_performance_monitor GL_EXT_texture_buffer_object GL_AMD_texture_texture4 GL_ARB_copy_buffer GL_ARB_depth_buffer_float GL_ARB_draw_instanced GL_ARB_half_float_vertex GL_ARB_instanced_arrays GL_ARB_map_buffer_range GL_ARB_texture_buffer_object GL_ARB_texture_rg GL_ARB_texture_swizzle GL_ARB_vertex_array_bgra GL_EXT_texture_swizzle GL_EXT_vertex_array_bgra GL_NV_conditional_render GL_AMD_conservative_depth GL_AMD_depth_clamp_separate GL_AMD_draw_buffers_blend GL_AMD_seamless_cubemap_per_texture GL_AMD_shader_stencil_export GL_ARB_ES2_compatibility GL_ARB_blend_func_extended GL_ARB_compatibility GL_ARB_debug_output GL_ARB_draw_buffers_blend GL_ARB_draw_elements_base_vertex GL_ARB_explicit_attrib_location GL_ARB_fragment_coord_conventions GL_ARB_provoking_vertex GL_ARB_sample_shading GL_ARB_sampler_objects GL_ARB_seamless_cube_map GL_ARB_shader_stencil_export GL_ARB_shader_texture_lod GL_ARB_tessellation_shader GL_ARB_texture_buffer_object_rgb32 GL_ARB_texture_cube_map_array GL_ARB_texture_gather GL_ARB_texture_multisample GL_ARB_texture_query_lod GL_ARB_texture_rgb10_a2ui GL_ARB_uniform_buffer_object GL_ARB_vertex_type_2_10_10_10_rev GL_EXT_provoking_vertex GL_EXT_texture_snorm GL_EXT_texture_storage GL_MESA_texture_signed_rgba GL_NV_copy_image GL_NV_texture_barrier GL_ARB_draw_indirect GL_ARB_get_program_binary GL_ARB_gpu_shader5 GL_ARB_gpu_shader_fp64 GL_ARB_robustness GL_ARB_separate_shader_objects GL_ARB_shader_bit_encoding GL_ARB_shader_precision GL_ARB_shader_subroutine GL_ARB_texture_compression_bptc GL_ARB_timer_query GL_ARB_transform_feedback2 GL_ARB_transform_feedback3 GL_ARB_vertex_attrib_64bit GL_ARB_viewport_array GL_EXT_direct_state_access GL_EXT_vertex_attrib_64bit GL_AMD_multi_draw_indirect GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_ARB_base_instance GL_ARB_compressed_texture_pixel_storage GL_ARB_conservati [... cut ...]
|    0   N/A  N/A      2992      G   /usr/bin/gnome-shell                           65MiB |
+-----------------------------------------------------------------------------------------+

The debugging information inside docker is not available, because such a problem when executing a `run.bash` statement will terminate the docker container running, and entering docker again will not retain any information. And I try to run `docker run-it osrf/subt-virtual-testbed:latest /bin/bash` will also report an error, because I am a docker beginner do not know how to further debug.

peci1, 2024-12-08:
The debugging values from inside Docker are important. Please, edit your `run.bash` and add the following after the line with `--privileged`:

    --entrypoint /bin/bash \

Then run just `./run.bash osrf/subt-virtual-testbed:latest` and it should throw you inside a bash prompt inside the Docker container. Then install the required files. You will most probably need to call `sudo apt update` before being able to install something.

jiaminguo1995, 2024-12-10:
> @peci1
>
>
> Hello, I am using a Docker image to start up, and when running ./run.bash osrf/subt-virtual-testbed competition.ign worldName:=tunnel_circuit_practice_01 circuit:=tunnel robotName1:=X1 [...]

There is no corresponding version of ROS1 for 24.04.  Could you please tell me how you solve this problem while using Subt?

kudouxiao, 2024-12-10:
> The debugging values from inside Docker are important. Please, edit your `run.bash` and add the following after the line with `--privileged`:
>
> ```
> --entrypoint /bin/bash \
> ```
>
> Then run just [...]

Thanks for your attention, I forgot to reply you. Finally, I decided to install one more ubuntu18.04 to start the project locally, which is more convenient for me to modify and deploy the project.

peci1, 2024-12-10:
> There is no corresponding version of ROS1 for 24.04. Could you please tell me how you solve this problem while using Subt?

There are 2 options:

1. Install Ubuntu 18.04 or Ubuntu 20.04
2. Use Docker

No other options seem possible (unless you want to help porting subt to ROS 2).

Option 1 comes with the problems that new hardware will have problems with the old operating systems. So it's best to run it on hardware from ~2020. Another option, as already mentioned in this issue, is installing mainline kernel and Mesa from kisak-mesa PPA. This could help on newer hardware, but it might be harder to maintain.